hku-mars/FAST_LIO #102: About memory usage

Psyclonus2887, 2021-12-04:
Thank you for your excellent work! The algorithm performs better than LIO_SAM in our dataset in off-road scenes, while I have some problem in memory usage. When using LIO-SAM, the memory usage is around 1%, but when using FAST-LIO2, memory usage is around 13.9%, in the same dataset around 200s. The pcd_save_en parameter is set to false, but it seems whether it is true or false, the usage of memory is equal. I also tried to adjust the cube_side_length parameter, it didn't make any difference. Is that a question led by the using of ikd-tree to organize local map?

XW-HKU, 2021-12-04:
Did you try to close the rviz, since the rviz will save all the points in the memory.

Psyclonus2887, 2021-12-04:
The memory usage is calculated when rviz is turned off. I use htop to check the mem usage of different thread

Psyclonus2887, 2021-12-04:
And the memory usage is growing when the map gets larger

XW-HKU, 2021-12-04:
I think the 10% - 20% consumption is normal and the local map size is constrained by the cube_side_length. Such that the memory consumption will grow to a certain point and now too much.

Psyclonus2887, 2021-12-04:
I agree that in theory the memory usage should be constrained by cube_side_length, but when I adjust this parameter, the mem usage just keep growing, and I am sure the map size is larger than the cube size

Psyclonus2887, 2021-12-18:
I output the size of kd-tree and map num when running the code, finding that the two number keeping growing through the whole procedure. I am sure the cube size is properly settled, it seems like the remove of map points is failed. Do you have any idea about it?

Psyclonus2887, 2021-12-18:
Another phenomenon is the kdtree_delete_counter is always 0

Ecstasy-EC, 2021-12-23:
@Psyclonus2887  
Sorry for the late reply. The number of valid tree nodes on ikd-Tree is maintained in the `valid_num()` which is very different from the tree size. Those deleted points are firstly deleted by attaching an tag on it ('deleted' label). They will be deleted during the rebuilding process when it is necessary. Further details could be achieved in our papaer. We also provide an example about this phenomenon in the ikd-Tree repo. 

The boxwise delete function will be trigged when you are moving close to the edge of the local map. We have tested this function in ikd-Tree repo for many times. But I am not sure if there is bug for this. If possible please provide more details or data for this.

Ecstasy-EC, 2021-12-30:
@Psyclonus2887 
Any updates about the memory usage?

Ecstasy-EC, 2022-01-04:
This issue is closed for no more updates. 
Thanks~

closed 2022-01-04 by Ecstasy-EC

jukindle, 2023-07-10:
@Ecstasy-EC I notice this issue, too. The memory usage just keeps growing even with the sensors static. After an hour or so, fast_lio crashes due to being out of memory on my machine with 64GB RAM. Would you have a suggestion how I could add a memory limit to the algorithm?

jukindle, 2023-07-10:
> > @Ecstasy-EC I notice this issue, too. The memory usage just keeps growing even with the sensors static. After an hour or so, fast_lio crashes due to being out of memory on my machine with 64GB RAM. Would you have a suggestion how I could add a memory limit to the algorithm?
> 
> Could you please have a look at the size of ikd-tree?

After 10 mins or so

tree size: 19440
tree valid num: 17432

jukindle, 2023-07-10:
x-axis is time [s], y-axis is number of entries [1] (red is the size of the kdtree, blue is the number of valid entries in the kdtree)
on average, the memory consumption of fast_lio grows at about 19MB/s. This plot was created with the sensor being static

[image]

jukindle, 2023-07-10:
@Ecstasy-EC I just realized that I was on the wrong trail, if I use your stock implementation my memory does not grow insanely as it does with ours. We branched out from fast lio lc which seems to be the cause of the issues. Will switch back to your standard implementation, thanks a lot for your help!

Ecstasy-EC, 2024-04-29:
> I have the same issue, but i could not understand how you solved it? @Ecstasy-EC 

Check the PCD save param in the config file first.

Btw, please open another issue and mention me there again if the solution above cannot work.