hku-mars/FAST_LIO #103: Logarithm of SO(3) has singularities

lewisjiang, 2021-12-05:
The logarithm map implementations in your code, namely 

https://github.com/hku-mars/FAST_LIO/blob/d0a877db69f569964697e1135e1fcb42aae9b3f2/include/Exp_mat.h#L75-L81
and 
https://github.com/hku-mars/FAST_LIO/blob/d0a877db69f569964697e1135e1fcb42aae9b3f2/include/so3_math.h#L81-L87

have singularities. If a rotation matrix has an eigen value equals to -1, the log is not unique. For example, if `R=[-1, 0, 0, 0, 1, 0, 0, 0, -1]`, your logarithm will map to `[0, 0, 0]`, while the correct answer is `[0, pi, 0]`. This is also mentioned in [wikipedia](https://en.wikipedia.org/wiki/Axis%E2%80%93angle_representation#Log_map_from_SO(3)_to_%F0%9D%94%B0%F0%9D%94%AC(3)).

XW-HKU, 2021-12-19:
Error-state usually assumed to be very small.

lewisjiang, 2021-12-19:
> Error-state usually assumed to be very small.

I see. But the `Log()` function also takes nominal states as input in your code.

XW-HKU, 2021-12-19:
Actually only implemented in error state when state estimation. You may some data log use this function but has,no influence in state estimation.

lewisjiang, 2021-12-21:
Thanks for the reply, Closing now...

closed 2021-12-21 by lewisjiang