hku-mars/FAST_LIO #105: 原地选转时，里程计“飞”了

gongyue666, 2021-12-11:
我用MID 70传感器和外置imu试验了该工程，在机器人原地旋转的时候，里程计大概率会“飞”，但log基本没有异常提示，如果不旋转，只是控制机器人来回前后做往复运动，则基本都是正常的。请问该问题如果可以改善？

gongyue666, 2021-12-11:
[image]

XW-HKU, 2021-12-11:
检查imu和雷达坐标轴有没有对齐，然后imu数据有没有丢帧。

gongyue666, 2021-12-11:
你好，坐标轴对齐如何验证呢？我标定以后也无法判断是否是正确的，用尺子大概量了一下，请问有更加准确的方法么？

gongyue666, 2021-12-11:
另外，我发现融合了imu的机械里程计在原地旋转时，位姿还是比较准的，是否有可能引入它们呢

XW-HKU, 2021-12-11:
坐标轴方向是不是对齐的？

gongyue666, 2021-12-11:
如何验证是否对齐？我安装打孔的时候尽量保证二者同轴，标定结果显示xyz三个轴的朝向偏移都在0.0x弧度这个级别

XW-HKU, 2021-12-11:
一帧一帧的看imu有没有丢帧。你这个现象大概率是imu通讯阻塞丢帧了或者lidar丢帧了

gongyue666, 2021-12-11:
好的，那我录制个包，检查一下

gongyue666, 2021-12-13:
[image]
你好，我录制了约一分钟的bag，查看时候发现有一段如上图所示的imu数据“空白”，不太确认是丢帧还是阻塞了

gongyue666, 2021-12-13:
imu的数据不是很均匀，这样会产生影响吗？

gongyue666, 2021-12-13:
另外imu和雷达的坐标轴对齐，是xyz的朝向保证一致就可以吧？

gongyue666, 2021-12-13:
发现了一种比较容易浮现的方法，将雷达对准一面墙壁拍摄，当看到里程计开始“抖动”比较厉害时候，再把雷达转回来，朝向正常的场景，里程计大概率无法再次稳定，而是继续震荡变大，然后彻底“飞”了

XW-HKU, 2021-12-13:
面对一面墙是退化场景啊

gaoxiang12, 2022-01-19:
FastLIO2的点-面残差机制容易导致它发散。
如果中间一个点云由于某种原因，没能得到准确pose，那么放到local map中，会导致local map中出现错误位置的点云。
而下一帧在寻找最近邻时，由于五个最近邻点的机制，容易导致最近邻的平面中，一部分落在正确的点云上，一部分落在错误的点云上，进而导致本帧也出现发散。
所以这样一个系统容易由于一帧错位导致后续的帧都发生错位，产生上图的情况。

添加关键帧机制或者给历史点云增加更高的权重可以改善这个问题。

ziv-lin, 2022-01-19:
@gaoxiang12 你的这种策略我们之前考虑过，在低速场景下也许是能用的，但是在高速场景下（如无人机高速飞行），你不得不相信前一帧刚刚加入地图的点云。
事实上，我们组也有考虑过怎么解决某一帧点云错误pose导致的后续问题，如：
1. 加入雷达滑动窗口: Liu, Zheng, and Fu Zhang. "Balm: Bundle adjustment for lidar mapping."
2. 是用概率地图的的方案:  Yuan, Chongjian, et al. "Efficient and Probabilistic Adaptive Voxel Mapping for Accurate Online 3D SLAM."
3. 加入更多视觉观测，如R2LIVE, R3LIVE系列
4. 其他若干在延方案

XW-HKU, 2022-01-19:
> FastLIO2的点-面残差机制容易导致它发散。 如果中间一个点云由于某种原因，没能得到准确pose，那么放到local map中，会导致local map中出现错误位置的点云。 而下一帧在寻找最近邻时，由于五个最近邻点的机制，容易导致最近邻的平面中，一部分落在正确的点云上，一部分落在错误的点云上，进而导致本帧也出现发散。 所以这样一个系统容易由于一帧错位导致后续的帧都发生错位，产生上图的情况。
> 
> 添加关键帧机制或者给历史点云增加更高的权重可以改善这个问题。

@gaoxiang12 这个issue里面楼主提到的，“面对一个大墙抖动发散”，原因是：雷达退化方向的位移不可观，而可观性是滤波器收敛的必要条件。长时间的不可观（退化）必定会带来ekf的发散。工程上可以添加退化检测来防止发散，但是也仅仅是防止发散，退化方向的位移理论上就解不出来。

point plane 测量模型，本身确实不是最精准的模型。这个的改进可以参考楼上林博提到的文献2。

infinitysky, 2022-03-27:
假设，我只是假设哈，
请问是否在开阔地方使用？我遇到了类似的问题， 我们使用的 avia ， 在室内我的角向量只要偏转一点点，立刻就飞到姥姥家
但是我在室外林地等开阔地方使用，就都没有问题

yxllnSLAM, 2022-04-15:
> 2\. Efficient and Probabilistic Adaptive Voxel Mapping for Accurate Online 3D SLAM

> @gaoxiang12 这个issue里面楼主提到的，“面对一个大墙抖动发散”，原因是：雷达退化方向的位移不可观，而可观性是滤波器收敛的必要条件。长时间的不可观（退化）必定会带来ekf的发散。工程上可以添加退化检测来防止发散，但是也仅仅是防止发散，退化方向的位移理论上就解不出来。
>
> point plane 测量模型，本身确实不是最精准的模型。这个的改进可以参考楼上林博提到的文献2。

你好，请问下目前的FAST LIO2中已经采用了文献2中的体素地图的表示方法吗？

closed 2023-01-01 by stale[bot]