hku-mars/FAST_LIO #108: IMU offset on  Livox Horizon and  loosing track

mohsenhadadi, 2021-12-19:
Hi everyone 
When I use Fast-Lio for Livox Horizon mounted at a 90 degrees perpendicular to my IMU's Y axis ,I encountered the following errors that I have attached. Please explain a little more about the parameters that I highlighted
, I only encounter such a problem when I change livox angle related to my IMU angle downwards among Y axis ,
anybody can help me solving this problem??

[image]
[image]
[image]

XW-HKU, 2021-12-19:
Why dont you use insider IMU of horizon and trouble yourself.

mohsenhadadi, 2021-12-19:
Thanks for your response, but even when I use livox internal IMU and turn livox downward around Y axis the same problem will occur

XW-HKU, 2021-12-19:
> Thanks for your response, but even when I use livox internal IMU and turn livox downward around Y axis the same problem will occur

make sure there is no degeneration when you turn lidar downward.

mohsenhadadi, 2021-12-19:
> 

What do you mean by degeneration?
Why when I change its location and direction due to my IMU in other axises rather than Y, it is fine and I don't have any error. But when misalignment is around Y and 90 degrees downward this problem happens , Can you please explain degeneration for me

XW-HKU, 2021-12-19:
> Thanks for your response, but even when I use livox internal IMU and turn livox downward around Y axis the same problem will occur

WHAT DO YOU MEAN "turn livox downward around Y" when you use internal imu ???????

mohsenhadadi, 2021-12-20:
Look I have tried 2 different setups:
1st: using horizon + Fast Lio+ external IMU

2nd: using horizon and its internal IMU + Fast Lio 

In both situation when livox is looking downward  from my multirotor I get errors and generated points are not in correct position
But if I install It in forward looking position on my multirotor every thing is fine

XW-HKU, 2021-12-20:
If lidar is facing only one big surface like ground. It cannot locate itself.

mohsenhadadi, 2021-12-21:
It was an interesting point 
Thank you so much for taking the time to help solve my problem

mohsenhadadi, 2021-12-21:
Excuse me, I had a few more questions, can you help me?
Why should scan_line in Horizon config. File be 6 while horizon is not a linear LiDAR and it has a wide non-repetitive pattern?
And what about other parameters in the mapping file except extrinsics , do we need really to change them?

XW-HKU, 2021-12-21:
> Excuse me, I had a few more questions, can you help me? Why should scan_line in Horizon config. File be 6 while horizon is not a linear LiDAR and it has a wide non-repetitive pattern? And what about other parameters in the mapping file except extrinsics , do we need really to change them?

Actually the line number is not used in the livox lidar handler function.

mohsenhadadi, 2021-12-25:
Thank you so much

closed 2021-12-25 by mohsenhadadi