hku-mars/FAST_LIO #109: No defination of h_v in h_share_model and the update_iterated_dyn_share_modified() function does not use h_v ?

Woodpecker0, 2021-12-20:
I wonder if i can change noise model of os1 lidar to make mapping result better, but I can't find h_v in fastlio2. 

Is it simplified in update_iterated_dyn_share_modified()? 

I would be very appreciate if there's any help.

Joanna-HE, 2021-12-20:
In the lasermapping.cpp, when the function “update _iterated
_dyn_share_modified” is called, a parameter named “LASER_POINT_COV” should
be input, this is the noise of the lidar measurements.

Woodpecker0, 2021-12-20:
> In the lasermapping.cpp, when the function “update _iterated _dyn_share_modified” is called, a parameter named “LASER_POINT_COV” should be input, this is the noise of the lidar measurements. Woodpecker0 ***@***.***>于2021年12月20日 周一上午10:55写道：
> […](#)

But h_v is not defined, would this noise parameter work correctly ?

Joanna-HE, 2021-12-20:
In fastlio2, h_v is simplified as an identity matrix, and all the
measurement noise are modeled as same values. If you want to use specific
formulation of h_v, and model the measurement noise as different values,
you can try the function “init_dyn_share” and “update_iterated_dyn_share”
in the esekfom.hpp. Or changed the covariance update procedure of EKF in
the update_iterated_dyn_share_modified. To the best of my knowledge, the
improvement of changing to a specific h_v is limited.

Woodpecker0, 2021-12-20:
@Joanna-HE Thanks for your advise, I will have a try.

closed 2021-12-20 by Woodpecker0