hku-mars/FAST_LIO #111: About ikd-tree

Psyclonus2887, 2021-12-21:
I found that the radius-search function is not embedded in your ikd-tree as a public function and the same function is available in pcl's kd-tree. Do you have plan to add the function in the future? Another question is, I notice the code in the nearest point judgement, there is possibility the knn search can return points less than five, even when the searching radius is set to INFINITY, which kind of makes me curious.

Huguet57, 2021-12-21:
I think they are about to release the radius-search function or is already released: https://github.com/hku-mars/ikd-Tree/commit/6bc08351d57fbd990c80e80377224296129acb4b.

Psyclonus2887, 2021-12-22:
Thanks, it seems it has been merged but just hasn't been released.

Psyclonus2887, 2021-12-22:
> I think they are about to release the radius-search function or is already released: [hku-mars/ikd-Tree@6bc0835](https://github.com/hku-mars/ikd-Tree/commit/6bc08351d57fbd990c80e80377224296129acb4b).

And BTW have you encountered the case that failing to delete points by PointBox? It seems when I run FAST_LIO2 the delete function doesn't work and the kd-tree size keeps growing.

Ecstasy-EC, 2021-12-23:
@Psyclonus2887 
Could you please provide me any bags or data that you observed those weird phenomenon about ikd-Tree?

Psyclonus2887, 2021-12-23:
> @Psyclonus2887 Could you please provide me any bags or data that you observed those weird phenomenon about ikd-Tree?

Thx, I will reply you tomorrow with some log

Psyclonus2887, 2021-12-27:
@Ecstasy-EC
I test FAST-LIO in the ros bag of nclt dataset you released. 
[image]
[image]
I show two screen prints here in different time of this bag when running the code, where:
cub_needrm size is the output of cub_needrm.size()
need remove cube number i is cub_needrm[i].vertex_min and cub_needrm[i].vertex_max
points deleted num is kdtree_delete_counter in your code
kdtree_size is ikdtree.size()

Ecstasy-EC, 2021-12-27:
> @Ecstasy-EC I test FAST-LIO in the ros bag of nclt dataset you released. [image] [image] I show two screen prints here in different time of this bag when running the code, where: cub_needrm size is [...]

The vertices of the boxes to be removed seem quite weird. Did you change the launch file and the yaml file? What is value of the param `cube_side_length` in the launch file and `DET_RANGE`?

Psyclonus2887, 2021-12-27:
cube_side_length is set to 20, and DET_RANGE is set to 100

Ecstasy-EC, 2021-12-27:
> cube_side_length is set to 20, and DET_RANGE is set to 100

Local map size(i.e., cube_size_length) should always be larger than the two times of the DET_RANGE. Pls refer to our paper about the working principle of local map management. Never change the parameters until you are sure how they work.

Psyclonus2887, 2021-12-27:
I will try it and read the paper again. Thanks a lot!!!

closed 2021-12-30 by Ecstasy-EC