hku-mars/FAST_LIO #112: Question about saving PCD files

hallyx, 2021-12-21:
I found that after 30min scanning by the Avia, the PCD file only records the last 10min the point cloud file. Is there anything to solve this problem？

XW-HKU, 2021-12-21:
I dont know

EloyAldao, 2022-01-14:
There is an option to divide the PCD output (-1 makes only one file, so can break) I think that it is in the config y'all, not in the launch file
I also split the bags I record every 2Gb to avoid corruption

PengKunPROO, 2022-11-17:
@EloyAldao Hi, I have a question about what value you set to the parameter "interval". If I set 10 to the interval,for which means I split the pcd file every 10 seconds?
Best wishes!waiting for your reply!

Tomato1107, 2022-11-17:
> @EloyAldao Hi, I have a question about what value you set to the parameter "interval". If I set 10 to the interval,for which means I split the pcd file every 10 seconds? Best wishes!waiting for your reply!

interval不是你认为的间隔时间，给interval赋大于1的值是指最后生成pcd的个数。看代码pcd应该是程序最后退出的时候才生成

closed 2022-11-17 by XW-HKU