hku-mars/FAST_LIO #113: Is Fast-LIO's Kinetic Model wrong?

Huguet57, 2021-12-21:
Hi all,
I have a worrying question about the **Kinetic Model**. I'm referring at the one referenced in the paper:

[image]

While trying to improve on the code, I have seen that integrating the IMUs (without LiDAR) works **too** bad. Here are the results:

- Ground truth circuit (with path drawn in green):
[image]

- Integrating IMUs with Fast-LIO's Kinetic Model starting from the yellow X:
[image]

Specially what worried me where the **curves**:
[image]
[image]

 What I thought is that Fast-LIO only considers the rotation to update the velocity and not the position. So if acceleration is 0 in a curve, Fast-LIO would not update the velocity and therefore neither the position.

Then, I tried to multiply the **rotation also on the position** (since integrating would leave the R, right?).

This is the result:

[image]

[image]

I put again the Ground Truth Path. Seems way more consistent, doesn't it?

Tell me your thoughts,
Andreu

Huguet57, 2021-12-21:
Additional screenshot of the **new proposed formulation** before closing the loop. It's really similar to the ground truth, it has the same curves.

[image]

With the **old formulation**:
[image]

Huguet57, 2021-12-21:
Second lap with new formulation, I think this is an obvious one. This is by integrating IMUs from the beginning alone.

[image]

Huguet57, 2021-12-21:
Either that or I'm understanding wrong the meaning the value GvI.
How I am doing the integration is:

Old:
`R    *= SO3::Exp(w, dt)`
`vel  += R*((imu.a - ba) + g)*dt`
`pos += vel*dt`

New:
`R    *= SO3::Exp(w, dt)`
`vel  += R*((imu.a - ba) + g)*dt`
`pos += R*vel*dt`

XW-HKU, 2021-12-22:
Thanks for your deep research.

I am a bit confused. The velocity in your new formular is also defined in world frame. If using the world frame velocity to update position, there should not have a rotation matrix multiplied in the left.

Huguet57, 2021-12-22:
@XW-HKU  Yes, that's what I was thinking too this morning.
I tried rotating the velocity vector too bit by bit:

`// R ⊞ (w - bw - nw)*dt`
`Eigen::Vector3f w_hat = imu.w - this->bw;`
`this->R *= SO3::Exp(w_hat, dt);`
`// v ⊞ (R*(a - ba - na) + g)*dt`
`this->vel = SO3::Exp(w_hat, dt)*this->vel; // too fast`
`this->vel = this->R*((imu.a - this->ba) + this->g)*dt; // assuming flat surface`
`// p ⊞ v*dt`
`this->pos += this->vel*dt;`

The results are also good but it accelerates way more...

Here is the first lap:
[image]

Here is the second lap:
[image]

This theorically makes more sense since I'm transforming the velocity vector the same way I'm transforming the states' rotation. But the velocities are more controlled in the previous method.

XW-HKU, 2021-12-22:
this->vel = this->R*((imu.a - this->ba) + this->g)*dt;

Huguet57, 2021-12-22:
Oh, this is actually another separate thing. I am assuming that the car is on a flat surface so it's better to compensate for gravity directly to the IMU reading than the rotated one. I haven't thought about putting the acceleration, I will try it.

Huguet57, 2021-12-23:
After studying hard the problem and thanks to external help and tips from my researcher I have concluded that indeed **there's nothing wrong** with Fast-LIO's model. While the "new formulation" being more aesthetically pleasing to the human eye, the below graphics talk for themselves.

The old formulation has a position error that keeps on increasing **consistently**. As expected with IMUs. The new, while seeming better to the naked eye, has an inconsistency to it that would not help when trying to model its error.

:heavy_check_mark: **Old** formulation:
- X global position error
[image]
- Y global position error
[image]

:x: **New** formulation:

- X global position error
[image]

- Y global position error
[image]
 
The "old formulation" is now the following (I tried use the previous values as done in a linear update):

`Eigen::Matrix3f Rtemp = R;`
`Eigen::Vector3f veltemp = vel;`
`Eigen::Vector3f postemp = pos;`

`Rtemp *= SO3::Exp(imu.w - bw, dt);`
`veltemp += (R*(imu.a - ba) - g)*dt;`
`postemp += vel*dt + 0.5*(R*(imu.a - ba) - g)*dt*dt;`

`R = Rtemp;`
`vel = veltemp;`
`pos = postemp;`

Thanks also @XW-HKU for the tips, I will keep trying to improve your amazing project for spinnings LiDARs. I am closing this.

closed 2021-12-23 by Huguet57

XW-HKU, 2021-12-23:
good