hku-mars/FAST_LIO #115: KITTI dataset error "Failed to find match for field 'time'."

GreatEms, 2021-12-30:
Thank you for your bravo contribution!
I tried the [http://www.cvlibs.net/datasets/kitti/raw_data.php](url) KITTI dataset with [https://github.com/tomas789/kitti2bag](url) tool to transfer the Synchronized IMU and velodyne to rosbag. I followed the Readme to edit yaml file, then played the kitti rosbag with error  "Failed to find match for field 'time'."  I wondered why Synchronized IMU failed, and how to run the KITTI rosbag properly.
I am really looking forward to your reply.

Huguet57, 2022-01-02:
This is an error of the ``POINT_CLOUD_REGISTER_POINT_STRUCT``, are you using this type?

``POINT_CLOUD_REGISTER_POINT_STRUCT(velodyne_ros::Point,``
``    (float, x, x)``
``    (float, y, y)``
``    (float, z, z)``
``    (float, intensity, intensity)``
``    (float, time, time)``
``    (std::uint16_t, ring, ring)``
``)``

Sunshinehaozi, 2022-05-10:
So far as I know,the data of HDL-64 in kitti dataset dose not have timestamp field

Shubham-2302, 2022-10-25:
@GreatEms Were you able to solve it ?

closed 2023-01-01 by stale[bot]

lijing137, 2024-04-18:
> @GreatEms Were you able to solve it ?

Have you solved this problem?

hankyeolyu, 2025-04-24:
Have you solved this problem?

Huguet57, 2025-04-24:
> This is an error of the `POINT_CLOUD_REGISTER_POINT_STRUCT`, are you using this type?
> 
> `POINT_CLOUD_REGISTER_POINT_STRUCT(velodyne_ros::Point,` ` (float, x, x)` ` (float, y, y)` ` (float, z, z)` ` (float, intensity, intensity)` ` (float, time, time)` ` (std::uint16_t, ring, ring)` `)`

It's probably this thing, guys