hku-mars/FAST_LIO #120: Add ROS2 support

ItsJas, 2022-01-27:
Hi,
Do you have any plans to port the package to ros2?

borongyuan, 2022-02-02:
Hi,
I'm porting this project to ros2.
https://github.com/borongyuan/FAST_LIO/tree/ros2
It's nearly finished but not yet tested.
Thanks

Tomato1107, 2022-02-03:
> I meet some compile errors on ros2 Foxy.
[image]

borongyuan, 2022-02-03:
Looks like these are just warnings, we'll see if we can get rid of them later.

XW-HKU, 2022-03-23:
> Hi, I'm porting this project to ros2. https://github.com/borongyuan/FAST_LIO/tree/ros2 It's nearly finished but not yet tested. Thanks

Thankyou very much！

Tomato1107, 2022-10-13:
> Looks like these are just warnings, we'll see if we can get rid of them later.

Finally I compile the ros2 version without errors, are there any lanuch.py files?

closed 2023-01-01 by stale[bot]