hku-mars/FAST_LIO #121: Livox Horizon closed loop mapping has drifting problem

DannyRaspect, 2022-01-31:
Location: Carpark

Lidar model: Livox horizon lidar ( just use built-in imu )

Record command: rosbag record -a 

I want to ask it is the slam config or hardware lend to the drift result?

[image]
[image]
[image]

kxch928925819, 2022-03-30:
你可以试着采用以人为轴心的左右摆动式扫描来试试！我也用horizon，地下车库是成果最后的地方！

liunao123, 2022-07-19:
很奇怪，程序启动的时候，如果一直往一个方向运动，轨迹的 Z 值会一直增大，大概 0.1m/s ，然后反方向回来的话，Z 值会逐渐减小，大概 -0.1m/s 。回到原点，差不多会变成0，范围越大，回到原点的误差就越大。请指点一下可能是啥原因，感谢。

closed 2023-01-01 by stale[bot]