hku-mars/FAST_LIO #122: What is the coordinate of the output pose?

shrazshad, 2022-02-03:
Is the pose in IMU frame or LiDAR frame?

Joanna-HE, 2022-02-04:
Pos is the position of IMU in the global frame.

closed 2022-02-11 by XW-HKU

ciwedwang, 2022-08-15:
what is global frame? first imu pose which align with gravity?

Joanna-HE, 2022-08-16:
The global frame is defined as the first IMU body frame when you start to record the rosbag. The IMU' direction relative to the gravity depends on how you put your IMU at the beginning of rosbag recording.
________________________________
寄件者: ciwedwang ***@***.***>
寄件日期: 2022年8月15日 下午 01:33
收件者: hku-mars/FAST_LIO ***@***.***>
副本: He Dongjiao ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/FAST_LIO] What is the coordinate of the output pose? (Issue #122)

what is global frame? first imu pose which align with gravity?

ciwedwang, 2022-08-16:
> 

Thank you for your replay, I found a strange question, 
R_imu_lidar: (lidar pich 5deg)
[ 0.9961947, 0.0000000, 0.0871557,
0.0000000, 1.0000000, 0.0000000,
-0.0871557, 0.0000000, 0.9961947 ]
t_imu_lidar: [ 1.717, -0.0, 1.138 ]

after scan measurement, state is:
state r: -0.000345237 00-0.0374697 0-0.00386671 000000.99929, t: 0.0423497 0.0939147 -1.02182, off r: 000000000 0.0436194 000000000 00.999048, t: 1.717 00000 1.138, ba: -1.03539e-06 -1.72492e-06 01.88017e-05, bg: -0.000181886 04.04175e-05 06.04094e-05 

state t : [0.0423497 0.0939147 -1.02182]
state euler ange : [ x: -0.0396451, y: -4.2947493, z: -0.0014865 ]

if i set R_imu_lidar as identity on purpose,
after scan measurement, state is:
state r: -0.00214257 00.00397316 00.00080698 0000.999989, t: 00.0879747 -0.0151787 -0.0282642, off r: 0 0 0 1, t: 1.717 00000 1.138, ba: -1.59745e-06 02.96747e-07 05.22169e-07, bg: -0.000144082 07.62581e-06 05.70473e-05

state t : [ 0.0879747 -0.0151787 -0.0282642]
state euler angel [ x: -0.2455266, y: 0.4552911, z: 0.0009755 ]

it seems first lidar frame is identity...

Joanna-HE, 2022-08-16:
Of course, the first frame of IMU should be identity. The estimation of the IMU direction is the rotation of IMU body frame relative to the global frame, i.e., the first IMU frame. The rotation of IMU's first frame to itself is exactly identity.
________________________________
寄件者: ciwedwang ***@***.***>
寄件日期: 2022年8月16日 下午 03:03
收件者: hku-mars/FAST_LIO ***@***.***>
副本: He Dongjiao ***@***.***>; Comment ***@***.***>
主旨: Re: [hku-mars/FAST_LIO] What is the coordinate of the output pose? (Issue #122)

Thank you for your replay, I found a strange question,
R_imu_lidar: (lidar pich 5deg)
[ 0.9961947, 0.0000000, 0.0871557,
0.0000000, 1.0000000, 0.0000000,
-0.0871557, 0.0000000, 0.9961947 ]
t_imu_lidar: [ 1.717, -0.0, 1.138 ]

after scan measurement, state is:
state r: -0.000345237 00-0.0374697 0-0.00386671 000000.99929, t: 0.0423497 0.0939147 -1.02182, off r: 000000000 0.0436194 000000000 00.999048, t: 1.717 00000 1.138, ba: -1.03539e-06 -1.72492e-06 01.88017e-05, bg: -0.000181886 04.04175e-05 06.04094e-05

state t : [0.0423497 0.0939147 -1.02182]
state euler ange : [ x: -0.0396451, y: -4.2947493, z: -0.0014865 ]

if i set R_imu_lidar as identity on purpose,
after scan measurement, state is:
state r: -0.00214257 00.00397316 00.00080698 0000.999989, t: 00.0879747 -0.0151787 -0.0282642, off r: 0 0 0 1, t: 1.717 00000 1.138, ba: -1.59745e-06 02.96747e-07 05.22169e-07, bg: -0.000144082 07.62581e-06 05.70473e-05

state t : [ 0.0879747 -0.0151787 -0.0282642]
state euler angel [ x: -0.2455266, y: 0.4552911, z: 0.0009755 ]

it seems first lidar frame is identity...

ciwedwang, 2022-08-17:
But imu frame is not identity, 
if  R_imu_lidar is identity, it seems normal, first imu frame is t: [ 0.0879747 -0.0151787 -0.0282642]  euler: [ x: -0.0396451, y: -4.2947493, z: -0.0014865 ]

if R_imu_lidar not identity,  lidar pitch angle is 5deg, first imu frame is t: [0.0423497 0.0939147 -1.02182]  euler: [ x: -0.0396451, y: -4.2947493, z: -0.0014865 ]