hku-mars/FAST_LIO #125: > 你现在就可以试试打开time_sync_en，看看mid70的效果如何。

lijian636, 2022-03-01:
> 你现在就可以试试打开time_sync_en，看看mid70的效果如何。

**Rosbag和录像在下面链接**：
https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D

**测试机体是mid-70** 
没有做imu和lidar的位置的准确标定，imu在lidar左半边上方如下图：
[image]

**代码能跑，墙体似乎比loam版本更平直，点云更清晰，还发现如下点**：

- x和y轴较稳定

- 扫描出的点云体有时轻微错位重叠，怀疑是lidar&imu sync没有做硬同步的原因

- z轴变动超过15度左右，会产生大范围漂移 (看视频第40s： ROSBAG和video.mp4 )

- 实际操作和z轴数据上下相反：往上抬高机器，rviz里雷达会往下扫；往下压机器，rviz里雷达会向上扫 (同样看第40s视频： ROSBAG和video.mp4 )。猜测可能是avia.yaml里，单位矩阵或Z做标符号填写的原因，如下图：
[image]

_Originally posted by @Noahcuptea in https://github.com/hku-mars/FAST_LIO/issues/25#issuecomment-876541461_

lijian636, 2022-03-01:
你好,请问你把mid70和imu做了时间同步就可以在rviz出结果吗

zfc-zfc, 2022-07-04:
> > 你现在就可以试试打开time_sync_en，看看mid70的效果如何。
> 
> **Rosbag和录像在下面链接**： https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D
> 
> **测试机体是mid-70** 没有做imu和lidar的位置的准确标定，imu在lidar左半边上方如下图： [image]
> 
> **代码能跑，墙体似乎比loam版本更平直，点云更清晰，还发现如下点**：
> 
> * x和y轴较稳定
> * 扫描出的点云体有时轻微错位重叠，怀疑是lidar&imu sync没有做硬同步的原因
> * z轴变动超过15度左右，会产生大范围漂移 (看视频第40s： ROSBAG和video.mp4 )
> * 实际操作和z轴数据上下相反：往上抬高机器，rviz里雷达会往下扫；往下压机器，rviz里雷达会向上扫 (同样看第40s视频： ROSBAG和video.mp4 )。猜测可能是avia.yaml里，单位矩阵或Z做标符号填写的原因，如下图：
>   [image]
> 
> _Originally posted by @Noahcuptea in [#25 (comment)](https://github.com/hku-mars/FAST_LIO/issues/25#issuecomment-876541461)_

FAST-LIO依赖于精确的雷达-IMU外参和精确的时间同步，如果没有做标定，仅仅是把IMU贴在雷达上面可能是无法得到好的结果的。对于LiDAR-IMU外参和时间标定，可以参考我们最近的工作 LI-Init: https://github.com/hku-mars/LiDAR_IMU_Init

closed 2023-01-01 by stale[bot]

jllk47, 2023-12-19:
> > 你现在就可以试试打开时间同步，看看70年代中期的效果如何。
> 
> **罗斯贝格和录像在下面链接**： https://www.jianguoyun.com/p/DQS2A9gQsP6nCBisvv8D
> 
> **测试机体是70年代中期**没有做（供烤肉或煮菜用的）浅炊坑和激光雷达的位置的准确标定，imu在激光雷达左半边上方如下图： [image]
> 
> **代码能跑,墙体似乎比壤土版本更平直,点云更清晰,还发现如下点**：
> 
> * x和y轴较稳定
> * 扫描出的点云体有时轻微错位重叠,怀疑是激光雷达和惯性测量单元同步没有做硬同步的原因
> * z轴变动超过15度左右,会产生大范围漂移（看视频第40年代:罗斯贝格和video.mp4)
> * 实际操作和z轴数据上下相反：往上抬高机器，rviz里雷达会往下扫；往下压机器，rviz里雷达会向上扫（同样看第40年代视频:罗斯贝格和video.mp4).猜测可能是avia.yaml里,单位矩阵或Z做标符号填写的原因,如下图：
>   [image]
> 
> _最初发布者@Noahcuptea在[#25(评论)](https://github.com/hku-mars/FAST_LIO/issues/25#issuecomment-876541461)_

您好，请问您关于imuz轴坐标与现实世界相反的问题解决了吗