hku-mars/FAST_LIO #128: Hi, I have an issue when using my Velodyne-VLP-16 lidar.

jhforstudy, 2022-03-03:
Hi, thanks for your great code.

I'm using ubuntu 18.04 & velodyne VLP-16 lidar.
I downloaded your package and rosbag file (I also used what you gave. HDL-32E rosbag file.)
I checked it has no problem with the rosbag file.
But, I tried to use my velodyne VLP-16 lidar instead of the rosbag file,

This error message comes:
"Failed to find match for field 'time'."

As you said, I think the timestamp is missing, but I don't know what to do.
I googled it and read velodyne manual too... but I failed.
Can you give me a help? I'm really looking forward to your answer.

arjunskumar, 2022-03-11:
@jhforstudy you need to set the point type of your lidar as XYZIRT

jhforstudy, 2022-03-11:
@arjunskumar Thank you for your answer! You mean that I have to publish XYZIRT, not XYZIR? Ok I’ll try it.

arjunskumar, 2022-03-11:
@jhforstudy yes

arjunskumar, 2022-08-08:
@jhforstudy are you able to fix the issue?

jhforstudy, 2022-08-09:
Yes @arjunskumar, my Velodyne topic doesn't have time stamp, that was the problem.
You can close this issue. Thanks.

MoonSheepJx, 2022-08-27:
> 

hello, i meet the same question about timestamp of velodyne16. 
I want to know velodyne16 simply cannot publish the timestamp of every point.
Or is it possible to set something of velodyne16 to enable the lidar to publish points with timestamp?

mteastepp, 2022-09-12:
@MoonSheepJx the velodyne driver that is installed using `apt install ros-x-velodyne` does not publish the point-wise timestamps. You need to pull the latest driver directly from the github page, then it will publish the point-wise timestamp.

closed 2023-01-01 by stale[bot]