hku-mars/FAST_LIO #130: Double checking frames

dotnoppler, 2022-03-07:
Hi, just wanted to double check:

Am I correct in the assumptions that the frames used by FAST_LIO of "body" and "camera_init" correspond to the ROS REP 105 frames "base_link" and "odom" respectively?

Thanks in advance!

closed 2023-01-01 by stale[bot]

Strinkin, 2024-04-13:
i have the same question

Gabryss, 2024-07-31:
I also have the same question

Strinkin, 2024-07-31:
> I also have the same question
我理清了他们之间的关系。
map -> camera_init -> body 
等同于
map -> odom -> base_link
或许我们可以保持联系，探讨3d建图和导航问题？

hi, I figured out the relationship between them.
map -> camera_init -> body 
is equal to
map -> odom -> base_link
Maybe we can keep in touch to discuss 3D mapping and navigation issues？

Gabryss, 2024-08-01:
Hi Strinkin,
Thank you for your response.
Sure let's keep in touch !

hermanndererdmann, 2024-08-05:
how can i link the map to camera_init?
when i run ros2 run tf2_ros tf2_echo camera_init map i get the error that no connection can be found because they are in two different trees.
any ideas?

Strinkin, 2024-08-05:
> how can i link the map to camera_init?
> when i run ros2 run tf2_ros tf2_echo camera_init map i get the error that no connection can be found because they are in two different trees.
> any ideas?

map通常与odom相连，或许你可以尝试把odom remap 成camera_init？

hermanndererdmann, 2024-08-05:
how would i do that?i tried with odom aswell no connection do body or camera_init