hku-mars/FAST_LIO #131: Loop back之后系统停止

XinZhaoDC, 2022-03-12:
我采用的是***avia***传感器，在测试数据时出现了loop back的报错，同时建图系统停止。请问这个问题应该怎么解决？
下面是相关截图：
[image]
[image]
[image]
[image]

XinZhaoDC, 2022-03-12:
[image]
[image]
[image]
[image]

XW-HKU, 2022-03-23:
大概率是IMU数据时间戳在某个时刻错乱了。

XinZhaoDC, 2022-03-28:
> 大概率是IMU数据时间戳在某个时刻错乱了。

谢谢，所以针对这种问题，有没有什么解决方法呢？只能换数据了么？

closed 2022-09-07 by Ecstasy-EC