hku-mars/FAST_LIO #132: Incorrect timestamp_unit for Velodyne config

kennyjchen, 2022-03-14:
In `config/velodyne.yaml`, `timestamp_unit` should be `0` since Velodyne pointwise timestamps are in seconds, not microseconds. Thanks.

Ecstasy-EC, 2022-03-15:
Thanks for your notice. The default time unit is set according to our tested datasets. You could modify it to fit your applications.

XW-HKU, 2022-03-17:
Different velodyn datesets have different time unit. You should carefully check it before you test any code or datesets.

kennyjchen, 2022-03-17:
Understood, thanks very much.

FYI for all -- the official ROS driver for Velodyne sensors timestamps points in units of seconds:
https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/lib/rawdata.cc#L655

closed 2022-03-17 by kennyjchen