hku-mars/FAST_LIO #133: Problem for using Robosense RS-Lidar-16 & IMU

overevil, 2022-03-15:
Thanks for the great work
When Im using Robosense RS-Lidar-16 & IMU there is someting wrong with the FAST-LIO. Two videos show the phenomenon.  They are running the same rosbag. One for Fast-LIO and another for LIO-SAM

overevil, 2022-03-15:
[https://v.youku.com/v_show/id_XNTg1MTAzODM0OA==.html?spm=a2hcb.profile.app.5~5!2~5~5!3~5!2~5~5~A](url)

this is Fast-Lio. Odometry is floating

overevil, 2022-03-15:
this is Fast-Lio. Odometry is floating

https://v.youku.com/v_show/id_XNTg1MTAzODM0OA==.html?spm=a2hcb.profile.app.5~5!2~5~5!3~5!2~5~5~A

overevil, 2022-03-15:
this is LIO-SAM. Odometry and Mapping is correct.

https://v.youku.com/v_show/id_XNTg1MDU0NDA2OA==.html?spm=a2hcb.profile.app.5~5!2~5~5!3~5!2~5~5!2~A

overevil, 2022-03-15:
This is my velodyne yaml. IMU and Lidar are in the same orientation and position. My pointcloud2 msgs has ring channel and time channel

common:
    lid_topic:  "/rslidar_points"
    imu_topic:  "/INS"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
    
preprocess:
    lidar_type: 2                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 16
    scan_rate: 20                # only need to be set for velodyne, unit: Hz,
    timestamp_unit: 0            # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 2

mapping:
    acc_cov: 8.2051057671444980e-03
    gyr_cov: 7.6675342143242911e-04
    b_acc_cov: 9.4891148211334220e-05
    b_gyr_cov: 3.8807514282326634e-05
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en:  true      # true: enable the online estimation of IMU-LiDAR extrinsic,
    extrinsic_T: [ 0.0,0.0,0.0]
    extrinsic_R: [ 1.0, 0.0, 0.0,
                   0.0, 1.0, 0.0,
                   0.0, 0.0, 1.0]

publish:
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

XW-HKU, 2022-03-15:
I think no need to change the values of acc_cov and gyr_cov.

overevil, 2022-03-15:
> I think no need to change the values of acc_cov and gyr_cov, because you didn't set the exact extrinsics.

I have tried the default acc_cov and gyr_cov settings. It does not work.

XW-HKU, 2022-03-15:
I looked the video, I think most of the lidar line lost. Since I didn't use RS-Lidar-16, I am not sure what's wrong, but you can check if the point number (variable pl_size in the [RS-Lidar-16](https://github.com/hku-mars/FAST_LIO/blob/ad692a628c8cb88c518d05e111b21b8974579098/src/preprocess.cpp#L192)) of each frame is right.

overevil, 2022-03-15:
> I looked the video, I think most of the lidar line lost. Since I didn't use RS-Lidar-16, I am not sure what's wrong, but you can check if the point number (variable pl_size in the [preprocess.cpp](https://github.com/hku-mars/FAST_LIO/blob/ad692a628c8cb88c518d05e111b21b8974579098/src/preprocess.cpp#L288)) of each frame is right.

Thanks I am checking it

overevil, 2022-03-29:
This cause the problem. Problem closed.

launch/mapping_velodyne.launch
<param name="point_filter_num" type="int" value="4"/>

closed 2022-03-29 by overevil

srinivasrama, 2022-05-18:
@overevil how you solved` Failed to find match for field 'time' `

overevil, 2022-05-18:
> @overevil how you solved`Failed to find match for field 'time'`

I modified the rslidar sdk and build point time as velodyne lidar. You can also find some ros pkgs to convert rslidar points to velodyne points

srinivasrama, 2022-05-18:
@overevil Thanks

Gatsby23, 2022-11-01:
> this is Fast-Lio. Odometry is floating
> 
> https://v.youku.com/v_show/id_XNTg1MTAzODM0OA==.html?spm=a2hcb.profile.app.5~5!2~5~5!3~5!2~5~5~A

Sorry for troubling you. I use the robosense lidar too, but the odometry drifts at once. Could you please share your experience about using this type of LiDAR？