hku-mars/FAST_LIO #135: Problem for init this project

infinitysky, 2022-03-19:
Hi all

I am trying to rebuild this project, and I already completed all the libraries installation process. 

However, I meet a problem with stage2 build:
```
 Build
Clone the repository and catkin_make:

    cd ~/$A_ROS_DIR$/src
    git clone https://github.com/hku-mars/FAST_LIO.git
    cd FAST_LIO
    git submodule update --init
    cd ../..
    catkin_make
    source devel/setup.bash
Remember to source the livox_ros_driver before build (follow 1.3 livox_ros_driver)
If you want to use a custom build of PCL, add the following line to ~/.bashrc export PCL_ROOT={CUSTOM_PCL_PATH}
```

when I running `git submodule update --init` commond I found this error messing

```
Warning: Permanently added the ECDSA host key for IP address '52.64.108.95' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:hku-mars/ikd-Tree.git' into submodule path '/home/ubuntu/Desktop/ws_livox/FAST_LIO/include/ikd-Tree' failed
Failed to clone 'include/ikd-Tree'. Retry scheduled
Cloning into '/home/ubuntu/Desktop/ws_livox/FAST_LIO/include/ikd-Tree'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:hku-mars/ikd-Tree.git' into submodule path '/home/ubuntu/Desktop/ws_livox/FAST_LIO/include/ikd-Tree' failed
Failed to clone 'include/ikd-Tree' a second time, aborting
```

is any one has idea to resolve it?

thank you

Poaos, 2022-03-23:
这个问题我也遇到过，你可以参考一下下图中的解决方案：**对FAST_LIO文件夹下面的隐藏文件.gitmodules进行修改，打开文件，把 git 形式的 url 改为 https 形式。**
[image]

除外，我把我之前测试fastlio的整个过程都记录在了下面这篇文章中，包括后面的 rosbag example，你都可以参考一下。
https://zhuanlan.zhihu.com/p/461246350

XW-HKU, 2022-03-23:
> 这个问题我也遇到过，你可以参考一下下图中的解决方案：**对FAST_LIO文件夹下面的隐藏文件.gitmodules进行修改，打开文件，把 git 形式的 url 改为 https 形式。** [image]
> 
> 除外，我把我之前测试fastlio的整个过程都记录在了下面这篇文章中，包括后面的 rosbag example，你都可以参考一下。 https://zhuanlan.zhihu.com/p/461246350

非常感谢！最新commit已经fix了这个问题。

closed 2022-03-23 by XW-HKU

infinitysky, 2022-03-23:
非常感谢几位的回复的帮助

我之前是直接删除 `include\ikd-Tree` 这个文件夹，然后直接在 `include` 这个文件夹项目做 git clone 完成的
感谢几位的时间和帮助，谢谢