hku-mars/FAST_LIO #136: Issue about The formula(18)

onionysy, 2022-03-21:
我在您的论文里看到，在卡尔曼滤波更新部分，您得出的状态更新公式（论文中的公式18）与一般的更新公式不同，能请问下您是怎么推导出来的吗 
[image]
?

XW-HKU, 2022-03-23:
这一块比较复杂，请参考我们的文章 https://arxiv.org/abs/2102.03804

inntoy, 2022-04-19:
> 你之前看到的常规卡尔曼滤波都是在笛卡尔坐标系下的实数域上推导的，如果系统状态中有旋转矩阵这种manifold上的状态，则update方程需要变化。这一块比较复杂，请参考我们的文章 https://arxiv.org/abs/2102.03804

您好，我看到你们引用的另一篇工作(LINS)里状态更新的公式和公式18不一致（它们也是迭代误差卡尔曼滤波）。然后我发现公式18广义加号右边括号里的那部分，其实就是误差状态的更新。这样公式18相比于LINS文章里的状态更新公式就有一个好处：迭代过程中误差状态的均值一直为0，这样每次更新和观测方程都可以在原点处线性化，请问我这样理解对吗？

inntoy, 2022-06-11:
@bugdeveloper0 抱歉，我最近在做其他的工作，已经有段时间没有看推导，相关公式忘的差不多了，可能没办法给你提供工作了。

HViktorTsoi, 2022-06-15:
> 我在您的论文里看到，在卡尔曼滤波更新部分，您得出的状态更新公式（论文中的公式18）与一般的更新公式不同，能请问下您是怎么推导出来的吗 [image] ?

请看 **Kalman Filters on Differentiable Manifolds** 的公式(38), **Performance evaluation of iterated extended Kalman filter with variable step-length**的公式(14, 19, 20)，以及**The Iterated Kalman Filter Update as a Gauss—Newton Method**的section V

closed 2022-06-18 by onionysy

inntoy, 2022-10-11:
您好，您的邮件我已收到。————祝您每天都有个好心情!