hku-mars/FAST_LIO #138: Pose6D.h problem during catkin_make

smileyenot983, 2022-03-23:
Hello everyone, i am trying to build the package, but when i write catkin_make it fails being unable to find Pose6D.h:

`[  8%] Building CXX object FAST_LIO/CMakeFiles/fastlio_mapping.dir/src/laserMapping.cpp.o
In file included from /home/ramil/catkin_ws/src/FAST_LIO/src/IMU_Processing.hpp:11:0,
                 from /home/ramil/catkin_ws/src/FAST_LIO/src/laserMapping.cpp:46:
/home/ramil/catkin_ws/src/FAST_LIO/include/common_lib.h:8:10: fatal error: fast_lio/Pose6D.h: No such file or directory
 #include <fast_lio/Pose6D.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.`

XW-HKU, 2022-03-23:
just build again, maybe need several times

smileyenot983, 2022-03-24:
After i build it for the second time i had same error, i have found discussions about creating custom ROS messages and found difference between their CMakeLists and this repo's. I have just added genmsg into find_package

`find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  nav_msgs
  sensor_msgs
  roscpp
  rospy
  std_msgs
  pcl_ros
  tf
  livox_ros_driver
  message_generation
  eigen_conversions
  genmsg
)`

and also added new line
`
add_dependencies(fastlio_mapping fast_lio_generate_messages_cpp)
`

That worked for me

infinitysky, 2022-04-03:
Please check your running environment
for me, I found this case caused by limited CPU core and RAM
I used running this process at VirtualBox with 1 CPU and 2 GB RAM, after I increased CPU core to 4 cores and 4GB RAM, then problem resolved.

srinivasrama, 2022-05-26:
try `catkin_make -j1`

SeanZsya, 2022-10-04:
> After i build it for the second time i had same error, i have found discussions about creating custom ROS messages and found difference between their CMakeLists and this repo's. I have just added [...]

After replacing "genmsg" with "message_generation" (which is already in the CMakeLists actually), this solution worked for me.

closed 2023-01-01 by stale[bot]

maripuusiim, 2023-03-04:
`CMakeFiles/Makefile2:531: recipe for target 'FAST_LIO/CMakeFiles/fastlio_mapping.dir/all' failed
`

Upgrading RAM from 4GB to 8GB solved issue for Ubuntu 18.04, ROS Melodic on VMware Fusion.

fanhuigui, 2023-04-04:
Hello, I have also encountered the same problem. After following your method, it still doesn't work

openhe-hub, 2023-05-20:
just run catkin_make again, this works for me

guchangjun, 2023-08-23:
I solved the problem:
add pose6D.h to the inclue file
the pose6D.h  can be downloaded in the path:
https://github.com/lus6-Jenny/MR_SLAM/blob/ce1ca08520f73183e48741406444418dd2264f00/Localization/devel/include/fast_lio/Pose6D.h#L4

sangshuduo, 2023-11-30:
> I solved the problem: add pose6D.h to the inclue file the pose6D.h can be downloaded in the path: https://github.com/lus6-Jenny/MR_SLAM/blob/ce1ca08520f73183e48741406444418dd2264f00/Localization/devel/include/fast_lio/Pose6D.h#L4

Thank you. This solution fixes the annoying build issue finally and is quite straightforward.

oscarkfpang, 2024-01-29:
I have the same problem of Pose6D.h during catkin_make of FAST-LIO2. My environment is Ubuntu 20 with ROS noetic. I'm using livox_ros_driver2 and livox_sdk2 for mid-360 lidar. I tried using `catkin_make -j1` but the build problem still existed.

I realised that I put FAST-LIO under the same workspace as the livox_ros_driver2. So I managed to solve the problem in this way

1. Separate the workspace for livox_ros_driver2 as the ros driver is built using its own `./build.sh ROS1` script. Source the livox_ros_driver2 workspace.
2. Put the FAST-LIO and POINT-LIO in another workspace folder and catkin_make as usual. 

I hope this helps other people having the same problem.

yuminhhuang, 2024-05-06:
> I have the same problem of Pose6D.h during catkin_make of FAST-LIO2. My environment is Ubuntu 20 with ROS noetic. I'm using livox_ros_driver2 and livox_sdk2 for mid-360 lidar. I tried using [...]

Great, thanks! This is exactly what I need.