hku-mars/FAST_LIO #142: 浙大lidar imu标定成果 如何才能写做velodyne yaml

kxch928925819, 2022-03-30:
[calib_result0328.csv](https://github.com/hku-mars/FAST_LIO/files/8377120/calib_result0328.csv)

mapping:
    acc_cov: 0.1
    gyr_cov: 0.1
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    fov_degree:    180
    det_range:     100.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic,
    extrinsic_T: [ 0, 0, 0.28]
    extrinsic_R: [ 1, 0, 0, 
                   0, 1, 0, 
                   0, 0, 1]

Ecstasy-EC, 2022-04-02:
读懂对应csv数据含义后，自行对应数据类型输入即可

zyx-99, 2022-04-21:
想问您的IMU是lord品牌的吗

Ecstasy-EC, 2022-04-22:
> 想问您的IMU是lord品牌的吗

不同数据集用的IMU不一样，请参考我们文章里给出的信息

zyx-99, 2022-04-22:
> > 想问您的IMU是lord品牌的吗
> 
> 不同数据集用的IMU不一样，请参考我们文章里给出的信息

十分感谢您的回复，我查看了您的文章，发现使用是Xsens M TIG 710的IMU，我现在手中的是与LIO SAM同款的IMU 3dm gx5 25
[image]
但此款IMU的XYZ轴对应的ROLL PITCH YAW貌似与Xsens不同（与雷达不对应），那是不是就无法跑通FAST-LIO呢。

Ecstasy-EC, 2022-04-22:
> > > 想问您的IMU是lord品牌的吗
> 
> > 
> 
> > 不同数据集用的IMU不一样，请参考我们文章里给出的信息
> 
> 
> 
> 十分感谢您的回复，我查看了您的文章，发现使用是Xsens M TIG 710的IMU，我现在手中的是与LIO SAM同款的IMU 3dm gx5 25
> 
> [image]
> 
> 但此款IMU的XYZ轴对应的ROLL PITCH YAW貌似与Xsens不同（与雷达不对应），那是不是就无法跑通FAST-LIO呢。

LIO-SAM的数据集我们是能跑的，但需要自行设置好外参。imu和雷达之间外参准确的情况下可以正常运行

zyx-99, 2022-04-29:
> 哈哈，我郁闷的有两个，第一 3dm-gx3-25的数据频率是100hz 我在琢磨给它改成200hz 第二 用浙大的计算lidar imu 结果 不太好 也就是标定的参数误差太大 第三 用了lidar align 标定 结果也不好 出现漂移很大 所以在考虑是不是问题在第一个上 数据频率太低  
> […](#)

方便留个邮箱交流一下嘛

kxch928925819, 2022-05-05:
***@***.*** 向你学习

  

  
方便留个邮箱交流一下嘛

kxch928925819, 2022-06-09:
名字中的数字就是QQ邮箱号码！还是没搞定，希望能得到你的指导啊！

zfc-zfc, 2022-07-04:
> 

雷达IMU外参标定可以参考我们的新工作 LI-Init: https://github.com/hku-mars/LiDAR_IMU_Init

closed 2022-09-07 by Ecstasy-EC

wangchuanhao1231, 2023-10-31:
> 名字中的数字就是QQ邮箱号码！还是没搞定，希望能得到你的指导啊！

请问你搞定了吗，我也在学习怎么把标定的结果写进yaml文件中。