hku-mars/FAST_LIO #144: 跑avia提供的数据集，rviz卡顿，为0FPS

kahowang, 2022-04-06:
本机配置 4800H + 32G ， 在跑 100hz_2021-02-05-19-50-54.bag  这个数据集时，跑了30S左右，rviz就卡顿明显，请问一下大佬是什么原因呢？

kahowang, 2022-04-06:
刚才跑了另一个提供的avia数据集，整体300S播放完成，rviz保持10FPS左右，看来是 “100hz_2021-02-05-19-50-54.bag ” 这个数据集有问题？

XW-HKU, 2022-04-06:
和数据集没关系，rviz本身就是卡，非常的卡

kahowang, 2022-04-06:
> 和数据集没关系，rviz本身就是卡，非常的卡

那请问一下大佬，平时调试的时候，不打开rviz调试？还是说只看轨迹？

XW-HKU, 2022-04-06:
绝大部分场合不需要100hz的刷新率，10hz rviz没有那么卡

kahowang, 2022-04-06:
> 绝大部分场合不需要100hz的刷新率，10hz rviz没有那么卡

好的，谢谢大佬，还有个问题就是，fastlio 中rviz显示的是全局地图，好像并没有对地图内存进行一些释放，导致地图越大，占用的内存越大，类似liosam，会在一段时间对部分内存进行释放。对显示的地图内存管理这方面，大佬觉得有需要优化的吗？

XW-HKU, 2022-04-06:
> > 绝大部分场合不需要100hz的刷新率，10hz rviz没有那么卡
> 
> 好的，谢谢大佬，还有个问题就是，fastlio 中rviz显示的是全局地图，好像并没有对地图内存进行一些释放，导致地图越大，占用的内存越大，类似liosam，会在一段时间对部分内存进行释放。对显示的地图内存管理这方面，大佬觉得有需要优化的吗？

rviz的地图内存什么的和fastlio无关。你可以自行修改rviz的地图持续显示时间。

kahowang, 2022-04-07:
好的，谢谢徐博指导~

Xu-Huitong, 2022-04-24:
> > 和数据集没关系，rviz本身就是卡，非常的卡
> 
> 那请问一下大佬，平时调试的时候，不打开rviz调试？还是说只看轨迹？

@XW-HKU 请问如果不开rviz的话，应该用什么进行实时的建图呢？或者说怎么样就可以避免卡顿现象呢：）

XW-HKU, 2022-04-24:
> > > 和数据集没关系，rviz本身就是卡，非常的卡
> > 
> > 
> > 那请问一下大佬，平时调试的时候，不打开rviz调试？还是说只看轨迹？
> 
> @XW-HKU 请问如果不开rviz的话，应该用什么进行实时的建图呢？或者说怎么样就可以避免卡顿现象呢：）

你的问题不是实时建图，你的问题是既要实时建图还要实时可视化建出来的几千万点云。可视化这个如果你觉得rviz卡可以去找其他插件或者自己开发可视化插件。

Xu-Huitong, 2022-04-24:
> 你的问题不是实时建图，你的问题是既要实时建图还要实时可视化建出来的几千万点云。可视化这个如果你觉得rviz卡可以去找其他插件或者自己开发可视化插件。

哦哦 原来如此，那徐博，请问您那边有推荐的插件吗：）

Xu-Huitong, 2022-04-24:
> > > > 和数据集没关系，rviz本身就是卡，非常的卡
> > > 
> > > 
> > > 那请问一下大佬，平时调试的时候，不打开rviz调试？还是说只看轨迹？
> > 
> > 
> > @XW-HKU 请问如果不开rviz的话，应该用什么进行实时的建图呢？或者说怎么样就可以避免卡顿现象呢：）
> 
> 你的问题不是实时建图，你的问题是既要实时建图还要实时可视化建出来的几千万点云。可视化这个如果你觉得rviz卡可以去找其他插件或者自己开发可视化插件。

另外，有点懵（尴尬）。请问您那边指的实时建图是什么含义呢：）不可视化点云如何建图呢：）

XW-HKU, 2022-04-24:
> > > > > 和数据集没关系，rviz本身就是卡，非常的卡
> > > > 
> > > > 
> > > > 那请问一下大佬，平时调试的时候，不打开rviz调试？还是说只看轨迹？
> > > 
> > > 
> > > @XW-HKU 请问如果不开rviz的话，应该用什么进行实时的建图呢？或者说怎么样就可以避免卡顿现象呢：）
> > 
> > 
> > 你的问题不是实时建图，你的问题是既要实时建图还要实时可视化建出来的几千万点云。可视化这个如果你觉得rviz卡可以去找其他插件或者自己开发可视化插件。
> 
> 另外，有点懵（尴尬）。请问您那边指的实时建图是什么含义呢：）不可视化点云如何建图呢：）

你不是做学术工作的吧？建图和可视化没有关系，图（map）是一种数据，给导航规划使用的数据。这个数据的功能是给上层导航算法调用。建图和可视化是完完全全两个事情。

Xu-Huitong, 2022-04-24:
> 

> 你不是做学术工作的吧？建图和可视化没有关系，图（map）是一种数据，给导航规划使用的数据。这个数据的功能是给上层导航算法调用。建图和可视化是完完全全两个事情。

哈哈 还是有点懵，也就是说实时建图是实时数据采集，实时可视化是数据的实时展示：）是这样吗
那请问您那边实时可视化通常是使用什么进行的呢：）就是应用fast-lio

XW-HKU, 2022-04-24:
> 哈哈 还是有点懵，也就是说实时建图是实时数据采集，实时可视化是数据的实时展示：）是这样吗 那请问您那边实时可视化通常是使用什么进行的呢：）就是应用fast-lio

你是做什么的？应用fastlio不需要实时可视化。

Xu-Huitong, 2022-04-24:
> 你是做什么的？应用fastlio不需要实时可视化。

我们想把框架部署在车上，然后车通过隧道，一边扫描同时实现对隧道的建模

closed 2022-04-24 by XW-HKU

fkdmmm, 2022-07-27:
试试录制rosbag

shallwing, 2025-02-19:
我之前也碰到了类似的问题，后面是通过降低pcl的版本得以解决的。pcl 1.12在运行时似乎占内存很大，独显的笔记本风扇转动声很响，而且在中途会出现运行崩溃的情况，后面使用pcl 1.8.1之后，既没有出现崩溃，风扇转动声也很小了。什么情况？