hku-mars/FAST_LIO #145: Is it work under low rate of lidar?

bigbigpark, 2022-04-11:
I have a ouster-32ch lidar. <br/>
I logged all data[lidar, imu, gps] under 1 Hz. <br/>

<br/>

When I run `roslaunch fast_lio mapping_ouster64.launch`, nothing worked on Rviz. <br/>
I have tested KITTI dataset (10Hz's Lidar), another dataset(maybe 20Hz's Lidar). I worked well <br/>

<br/>

The 1 Hz is too low to run this fast-lio2 ? <br/>
<br/>
Best regards.

closed 2022-04-28 by bigbigpark

Shubham-2302, 2022-10-25:
I am trying to use this algorithm for Kitti Dataset. Can you tell me how did you apply FastLio with Kitti Dataset
I am getting these errors
Failed to find match for field 'intensity'.
Failed to find match for field 'time'.
Failed to find match for field 'ring'.

JadrianChenson, 2025-01-20:
> I am trying to use this algorithm for Kitti Dataset. Can you tell me how did you apply FastLio with Kitti Dataset I am getting these errors Failed to find match for field 'intensity'. Failed to find match for field 'time'. Failed to find match for field 'ring'.

Hello, have you resolved these issues?