hku-mars/FAST_LIO #153: IMU outliers drift?

YangSiri, 2022-04-27:
Hi, DALAOMEN

I'm testing fast_lio with VLP16 and mti-300. However, it seems there are some outliers in the imu data. But i'm not so sure as the beginner. What would cause the tremendous drift in the pic? It happens at specific time without lack of imu data. 

scene1 at 64.7s
[image]

scene2 at 16.4s
[image]

Looking forward to your help~

kxch928925819, 2022-04-28:
你这貌似是imu标定的有问题 和我一个毛病 我在考虑的是： 第一、 是不是数据输出频率太低 目前出错是100hz 想改为200hz 。
第二，lidar imu标定的参数误差较大 考虑再次标定一下。 我也被标定结果困扰的不行，没有找到最恰当的解决方法！

YangSiri, 2022-05-01:
Thanks, @kxch928925819 

First, my IMU data is 200hz though with some of missings at full seconds.
Second, since the system runs for a while with a good mapping result, i dont think it is the extrinsic that causes this drift.

kxch928925819, 2022-05-05:
谢谢！找了一台win xp版本的电脑，使用官网程序，设置为200hz，问题解决了！

kxch928925819, 2022-05-05:
还是标定结果的问题，我的水平有点低，标定做了很多次，还是不能在lio-sam和fast-lio取得好成果，如果你有好经营，希望能分享，中午又去标定了一下，有好结果我会告诉你！ 共勉！

YangSiri, 2022-05-05:
Thannk you, @kxch928925819 

I have confirmed that this phenomenon is caused by large values in IMU raw data. So I wonder how to deal with it in fast_lio. 
BTW, the extrinsic of my system is measured by hand.

kxch928925819, 2022-05-06:
src/config/*.yaml

closed 2023-01-01 by stale[bot]

pjs9115916, 2023-03-24:
> 还是标定结果的问题，我的水平有点低，标定做了很多次，还是不能在lio-sam和fast-lio取得好成果，如果你有好经营，希望能分享，中午又去标定了一下，有好结果我会告诉你！ 共勉！
> […](#)

get the extrinsic from measurement by cad is better then align