hku-mars/FAST_LIO #155: IMU z Axis drift in a closed loop scan

mohsenhadadi, 2022-05-08:
Hello friends,
I have a livox horizon integrated with jetson xavier nx and an adis16470 imu data from " CUAV X7 Pro"  ,
and using PTP time synchronizing method.
Problem 1 : when scanner installed on a car roof and scanning a level area I have no problem and everything is fine
BUT the problem is when I scan an area with different elevations  and in the result point cloud and  the start points do not Mach the end points and it seems like it had a drift in z axis or not sensing street elevation correctly.
How can I fix it? is it IMU problem or synchronizing problem?

thank you for your answering...

rlabs-oss, 2022-06-26:
This is normal, FAST_LIO does not perform loop closure.

rlabs-oss, 2022-08-04:
normal problem due to drift - fast_lio does not loop close.

closed 2023-01-01 by stale[bot]