hku-mars/FAST_LIO #156: IMU suggestion?

rahulsharma11, 2022-05-09:
Hi,
I have been using Fast_LIO with Livox Avia for my 3D scanning project and it is very good.
Now i want to integrate external IMU for better longer run accuracy. I can use any of below-
1) Livox Avia
2) Velodyne puck

I have few options for selecting IMU-
1) MTi-610 (https://store.clearpathrobotics.com/products/mti-610)
2) NGIMU (https://x-io.co.uk/ngimu/)
3) WitMotion IMU (https://www.wit-motion.com/digital-inclinometer/witmotion-hwt905-rs485-high-accuracy.html)

Any suggestion for which IMU should I go in term of accuracy and support? 
Or any other IMU suggestion will also be considered.

Thanks.

rahulsharma11, 2022-05-13:
Hi any suggestion?

shikharx06, 2022-06-14:
Hi, I am also waiting for the reply for same issue.

aditdoshi333, 2022-06-14:
@XW-HKU please advise on the same. Even I am waiting to use FASTLIO with velodyne. 

Any light would be helpful

zfc-zfc, 2022-07-04:
The quality and accuracy of IMU are important, but the extrinsic and synchronization of LiDAR-IMU sometimes are more significant. For LiDARs without internal IMU, our recent work LI-Init (https://github.com/hku-mars/LiDAR_IMU_Init) is recommended.

rahulsharma11, 2022-07-04:
> The quality and accuracy of IMU are important, but the extrinsic and synchronization of LiDAR-IMU sometimes are more significant. For LiDARs without internal IMU, our recent work LI-Init (https://github.com/hku-mars/LiDAR_IMU_Init) is recommended.

Hi thanks for the wonderful work. I will give this a try.
Apart from that , my original question was which IMU will you prefer to use? I provided few options as well. Any suggestions on that? 
Thanks.

zfc-zfc, 2022-07-04:
> Hi thanks for the wonderful work. I will give this a try. Apart from that , my original question was which IMU will you prefer to use? I provided few options as well. Any suggestions on that? Thanks.

I didn't try the options you mentioned, so I cannot tell which one is the best. But it is recommended to read the datasheet of each product, the IMU with low-level noise and bias might be good.

closed 2023-01-01 by stale[bot]