hku-mars/FAST_LIO #159: start from desired pose

yogeshflux, 2022-05-11:
Hi team,

Thanks for the stack. I am learning a lot from it.

But I am trying to start the odometry to run from my own desired position instead of (0,0,0) and orientation also, tried to break the code and insert pose but it didn't work for me. Can you please help me in doing so? Thanks in advance

cate1717, 2022-12-01:
Hi @yogeshflux 
I have the same issue, did you resolve this problem in some way? there are any lines in the code where i can set the pose?
thank you in advance

yogeshflux, 2022-12-02:
Hello @cate1717 ,

Yes, I found that but it is affecting the state point estimation performance. Delay will cause and it might affect at the long run.
After iterated state estimation (line 955), you can you add you pose to transform.
ex: 

           // 955 :  
            state_point = kf.get_x();
 
            // added lines
            if (set_init_pose) //to check for desired pose is set.
            {
            
               //state_point pose into posestamp conversion
                geometry_msgs::PoseStamped pose_1;
                pose_1.pose.position.x = state_point.pos(0);
                pose_1.pose.position.y = state_point.pos(1);
                pose_1.pose.position.z = state_point.pos(2);
                pose_1.pose.orientation.x = state_point.rot.coeffs()[0];
                pose_1.pose.orientation.y = state_point.rot.coeffs()[1];
                pose_1.pose.orientation.z = state_point.rot.coeffs()[2];
                pose_1.pose.orientation.w = state_point.rot.coeffs()[3];
                
              //desired pose to use in transforming odometry in translation and rotation
                geometry_msgs::TransformStamped t;
                t.transform.translation.x = init_pose.pose.position.x;
                t.transform.translation.y = init_pose.pose.position.y;
                // t.transform.translation.z = init_pose.pose.position.z;

                tf::Quaternion q;
                q[0] =init_pose.pose.orientation.x;
                q[1] = init_pose.pose.orientation.y;
                q[2] = init_pose.pose.orientation.z;
                q[3] = init_pose.pose.orientation.w;
                tf::Matrix3x3 m(q);
                m.getRPY(offset_roll, offset_pitch, offset_yaw);
                tf::Quaternion nq;
                nq.setRPY(0, 0, offset_yaw);

                t.transform.rotation.x = nq[0];
                t.transform.rotation.y = nq[1];
                t.transform.rotation.z = nq[2];
                t.transform.rotation.w = nq[3];

                //transforming state point(pose_1) with desired pose (t) and putting back in state_point
                tf2::doTransform(pose_1, pose_1, t);

                state_point.pos(0) = pose_1.pose.position.x;
                state_point.pos(1) = pose_1.pose.position.y;
                state_point.pos(2) = pose_1.pose.position.z;
                state_point.rot.coeffs()[0] = pose_1.pose.orientation.x;
                state_point.rot.coeffs()[1] = pose_1.pose.orientation.y;
                state_point.rot.coeffs()[2] = pose_1.pose.orientation.z;
                state_point.rot.coeffs()[3] = pose_1.pose.orientation.w;
            }
            // end
            euler_cur = SO3ToEuler(state_point.rot);

Note : code isn't complete, and it is just a try out for an experiment. For better results you can use  [fast_lio_localization](https://github.com/HViktorTsoi/FAST_LIO_LOCALIZATION)

closed 2022-12-02 by yogeshflux