hku-mars/FAST_LIO #160: Measurement Jacobian matrix H

patripfr, 2022-05-13:
Hi,
thanks a lot for sharing this great framework.
I'm trying to understand the computation of the measurement jacobian.

From this it follows that `point_this` is the laser point expressed in the **LiDAR frame**:
```
V3D point_this = s.offset_R_L_I * point_this_be + s.offset_T_L_I;
M3D point_crossmat;
point_crossmat<<SKEW_SYM_MATRX(point_this);
```

From this it follows that `C` is the normal vector expressed in the **body frame**:
```
V3D norm_vec(norm_p.x, norm_p.y, norm_p.z);
 /*** calculate the Measuremnt Jacobian matrix H ***/
V3D C(s.rot.conjugate() *norm_vec);
```

`A` is then calculated as: 
`V3D A(point_crossmat * C);`

This means you are taking the cross product between `point_this` (which is the point expressed in  **LiDAR frame**) and `C` (which is the normal expressed in  **body frame**).
I don't quite understand why you take the cross product between vectors expressed in different frames?
I would be happy if you could explain this calculation or if I'm misunderstanding what `point_this` and `C` represent.

WFram, 2022-05-14:
Hi, 
Since `s.offset_R_L_I` and `s.offset_T_L_I` are LiDAR->IMU extrinsic (rotation matrix and translation vector expressed in IMU frame) from the state vector, it means that `point_this` is expressed in IMU frame and the measurement model is written correctly.

patripfr, 2022-05-16:
Thanks for pointing this out @WFram. 
I was assuming that `body` in `feats_down_body` represents the IMU frame, as this is the case in published odometry frame and tf, but apparently `body` in this case represents the LiDAR frame.

closed 2022-05-16 by patripfr