hku-mars/FAST_LIO #162: 宏LASER_POINT_COV在ESEKF更新过程中有什么作用?

liang0724s, 2022-05-21:
#define LASER_POINT_COV     (0.001)
kf.update_iterated_dyn_share_modified(**LASER_POINT_COV**, solve_H_time);

WFram, 2022-05-21:
`LASER_POINT_COV` is a covariance for LiDAR measurements and participates in calculating Kalman gain process which is essential to update the state vector.

liang0724s, 2022-05-21:
> `LASER_POINT_COV` is a covariance for LiDAR measurements and participates in calculating Kalman gain process which is essential to update the state vector.

Thanks, and how to get this value? is it from the LiDAR  user manual?

WFram, 2022-05-21:
There are many factors affecting measurement process, so in general this information can't be provided by only one value. Ideally, you need to take into account each individual measurement. And Livox Custom Messages actually provide this information in tag field (see https://www.livoxtech.com/showcase/livox-tag), but I would say, it's about confidence in general, rather than specific values. Even so, you can neglect this and approximate by one number and apparently it will not affect accuracy too much. But will increase performance significantly, since you don't need to calculate high-dimensional matrices and use only a scalar value.

closed 2022-05-23 by liang0724s