hku-mars/FAST_LIO #163: FASTLIO 中，状态方程的雅克比推导，A阵的推导求解惑

kahowang, 2022-05-23:
如下图1为高翔老师在简明ESKF中的状态方程F雅克比矩阵推导，图二为FASTLIO1 论文中的F阵推导，经比对，其他项的雅克比都是一样的，但是红色框的部分(delta_R 对 dleta_bg 求雅克比)，这里的A阵是一个小量吗，推导是怎样的，希望能解惑，谢谢。

[image]

[image]

kahowang, 2022-05-25:
> […](#)

非常感谢你的回复，想请问一下，这项是Jr吗？所以推导的运动模型也是和高博一样的是吗

Joanna-HE, 2022-10-11:
是Jr，和高博一样。

closed 2023-01-01 by stale[bot]