hku-mars/FAST_LIO #169: 请教参数的意义和调节方法

245680, 2022-06-18:
fov_degree,point_filter_num,max_iteration,filter_size_surf,filter_size_map,cube_side_length,这些参数是什么意思

[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow,在室内跑的时候报这个错怎么解决,应该调哪个参数

zfc-zfc, 2022-07-04:
> filter_size_surf,filter_size_map

室内可以把filter_size_surf,filter_size_map调小一点，一般分别为0.05, 0.15. 对于ouster或者这种点特别多的，point_filter_num可以调大，比如5~10.

plumewind, 2022-08-01:
借楼一用，谢谢。
知道参数cube_side_length是表示地图的局部区域的长度，而参数det_range表示激光雷达的最远探测距离。一般来说cube_side_length是det_range的两倍左右？是这样吗？

但是我现在遇到一个奇怪的现象，在隧道场景下，只要cube_side_length=det_range就能跑，cube_side_length!=det_range就不行，这是为什么呢？

hr2894235132, 2022-10-26:
> > filter_size_surf,filter_size_map
> 
> 室内可以把filter_size_surf,filter_size_map调小一点，一般分别为0.05, 0.15. 对于ouster或者这种点特别多的，point_filter_num可以调大，比如5~10.

您好，想请问一下这个fov为什么是90呢？我查看了avia的手册，它的fov是70.4° （水平）× 77.2°（竖直），我如果使用mid-70（它的FOV：70.4°圆形），config文件中的fov应该改成70吗？或者更小呢？可以给我点建议吗？

closed 2023-01-01 by stale[bot]

sumedhfb, 2024-07-11:
Any updates on this? Can someone help me with the significance and use of these parameters