hku-mars/FAST_LIO #170: Process has dies during playing the data

knurgazy, 2022-06-22:
I am using ouster lidar os1-128. When I use "rosbag play my_file.bag", I face this problem. 
`[laserMapping-2] process has died [pid 38057, exit code -11, cmd /home/ubuntu/A_ROS_DIR/devel/lib/fast_lio/fastlio_mapping __name:=laserMapping __log:=/home/ubuntu/.ros/log/54920bae-f201-11ec-9999-6d259b2401ed/laserMapping-2.log].
log file: /home/ubuntu/.ros/log/54920bae-f201-11ec-9999-6d259b2401ed/laserMapping-2*.log
`

closed 2023-01-01 by stale[bot]

shahaizi0, 2023-03-22:
请问解决了吗？我也遇到了这个问题 @knurgazy

huangyoupingben, 2023-04-18:
ikd_tree 初始化后，在插入点云的函数 add_points 时候，程序出现崩溃。点云已经做过异常值检查，没有nan 和 inf 等异常情况。 gdb捕获到的信息，如下图所示：
[image]
[image]

autel-liuxiaozi, 2024-07-01:
> I am using ouster lidar os1-128. When I use "rosbag play my_file.bag", I face this problem. `[laserMapping-2] process has died [pid 38057, exit code -11, cmd [...]

solved？this problem also occured in my situation

autel-liuxiaozi, 2024-07-01:
> 请问解决了吗？我也遇到了这个问题 @knurgazy

我也遇到了，请问怎么解决

autel-liuxiaozi, 2024-07-02:
解决了，源代码没有这个问题。但我们是修改过的代码所以有问题。将pcd点云输出到了其他路径，然后却又没有创建那个目录，导致出错

yuafowo, 2024-11-12:
> 解决了，源代码没有这个问题。但我们是修改过的代码所以有问题。将pcd点云输出到了其他路径，然后却又没有创建那个目录，导致出错

你好 我想请教一下 适配ouster128的时候，总是出现漂移的情况，还有''[WARN][1731321809.915177231]: No point, skip this scan!",你有遇到过这种情况嘛

hzlbbfrog, 2024-12-06:
> > 解决了，源代码没有这个问题。但我们是修改过的代码所以有问题。将pcd点云输出到了其他路径，然后却又没有创建那个目录，导致出错
> 
> 你好 我想请教一下 适配ouster128的时候，总是出现漂移的情况，还有''[WARN][1731321809.915177231]: No point, skip this scan!",你有遇到过这种情况嘛

你好，请问你解决了吗，我也是漂移然后提示这个问题。我是ouster64，我怀疑是config的问题，但是不知道正确的config呢

yuafowo, 2024-12-11:
> > > 解决了，源代码没有这个问题。但我们是修改过的代码所以有问题。将pcd点云输出到了其他路径，然后却又没有创建那个目录，导致出错
> > 
> > 
> > 你好 我想请教一下 适配ouster128的时候，总是出现漂移的情况，还有''[WARN][1731321809.915177231]: No point, skip this scan!",你有遇到过这种情况嘛
> 
> 你好，请问你解决了吗，我也是漂移然后提示这个问题。我是ouster64，我怀疑是config的问题，但是不知道正确的config呢

解决了在preprocess.cpp文件的ouster_handler函数中过滤掉nan点，即在循环中添加“if(isnan(pl_orig.points[i].x) || isnan(pl_orig.points[i].y) || isnan(pl_orig.points[i].z)) continue;” 希望可以帮助到你。

Ratarang, 2025-01-15:
> ikd_tree 初始化后，在插入点云的函数 add_points 时候，程序出现崩溃。点云已经做过异常值检查，没有nan 和 inf 等异常情况。 gdb捕获到的信息，如下图所示： [image] [image]

Hey bro, have you fixed your mentioned problem?