hku-mars/FAST_LIO #175: What is the relationship between the curvature of the point cloud and time？

LiquoriceH, 2022-07-25:
Hi，
in the func of “**bool sync_packages(MeasureGroup &meas)**” ，
lidar_end_time = meas.lidar_beg_time + **meas.lidar->points.back().curvature / double(1000)**;
Why curvature divided by 1000 is the interval time？

XW-HKU, 2022-07-25:
the unit of curvature is ms, not s.

closed 2022-07-25 by XW-HKU

LiquoriceH, 2022-07-25:
> the unit of curvature is ms, not s.

Thank for your explaination.
But I mean that why the unit of curvature is time? It should be m^-1.

XW-HKU, 2022-07-25:
> > the unit of curvature is ms, not s.
> 
> Thank for your explaination. But I mean that why the unit of curvature is time? It should be m^-1.

I just reuse this variable.

LiquoriceH, 2022-07-25:
> > > the unit of curvature is ms, not s.
> > 
> > 
> > Thank for your explaination. But I mean that why the unit of curvature is time? It should be m^-1.
> 
> I just reuse this variable.

Can you explain it in detail? I still don't understand why the curvature divided by 1000 is the scan interval time, I don't think curvature and time are related.
Thank you very much!

XW-HKU, 2022-07-25:
> Can you explain it in detail? I still don't understand why the curvature divided by 1000 is the scan interval time, I don't think curvature and time are related. Thank you very much!

I just use the curvature variable to save the timestamp, please refer to the preprocess.cpp to check how the curvature is used to save the timestamp.

LiquoriceH, 2022-07-25:
> I just use the curvature variable to save the timestamp, please refer to the preprocess.cpp to check how the curvature is used to save the timestamp.

Oh，I see. Thank you for your patience!

LiquoriceH, 2022-10-11:
Thank for your explaination.
But I mean that why the unit of curvature is time? It should be m^-1.