hku-mars/FAST_LIO #183: ERROR: cannot launch node of type

Nationarity, 2022-09-05:
Hello.

I am running the following, but it is not working.

Can someone helps me please.

user@ubuntu:~$ cd ~/catkin_ws/src
user@ubuntu:~/catkin_ws/src$ source ~/catkin_ws/devel/setup.bash
user@ubuntu:~/catkin_ws/src$ roslaunch fast_lio mapping_avia.launch

…
ERROR: cannot launch node of type [fast_lio/fastlio_mapping]: Cannot locate node of type [fastlio_mapping] in package [fast_lio]. Make sure file exists in package path and permission is set to executable (chmod +x)
…

I apologize. But what did I do wrong?

tohax, 2022-09-05:
user@ubuntu:~$ cd /catkin_ws
user@ubuntu:/catkin_ws$ source /catkin_ws/devel/setup.bash
user@ubuntu:/catkin_ws$ roslaunch fast_lio mapping_avia.launch

You need to source from /catkin_ws directory, not /catkin_ws/src

Nationarity, 2022-09-06:
>Thank you for your response.
  However, I get the same error.

user@ubuntu:~$ cd ~/catkin_ws
user@ubuntu:~/catkin_ws$ source devel/setup.bash
user@ubuntu:~/catkin_ws$ roslaunch fast_lio mapping_avia.launch
... logging to /home/user/.ros/log/3430ee0e-2d7d-11ed-b32c-907841d4d92e/roslaunch-ubuntu-10634.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:44153/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /livox/imu
 * /common/lid_topic: /livox/lidar
 * /common/time_offset_lidar_to_imu: 0.0
 * /common/time_sync_en: False
 * /cube_side_length: 1000.0
 * /feature_extract_enable: False
 * /filter_size_map: 0.5
 * /filter_size_surf: 0.5
 * /mapping/acc_cov: 0.1
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 450.0
 * /mapping/extrinsic_R: [1, 0, 0, 0, 1, 0...
 * /mapping/extrinsic_T: [0.04165, 0.02326...
 * /mapping/extrinsic_est_en: False
 * /mapping/fov_degree: 90
 * /mapping/gyr_cov: 0.1
 * /max_iteration: 3
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: True
 * /point_filter_num: 3
 * /preprocess/blind: 4
 * /preprocess/lidar_type: 1
 * /preprocess/scan_line: 6
 * /publish/dense_publish_en: True
 * /publish/path_en: False
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /runtime_pos_log_enable: False

NODES
  /
    laserMapping (fast_lio/fastlio_mapping)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [10645]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3430ee0e-2d7d-11ed-b32c-907841d4d92e
process[rosout-1]: started with pid [10656]
started core service [/rosout]
ERROR: cannot launch node of type [fast_lio/fastlio_mapping]: Cannot locate node of type [fastlio_mapping] in package [fast_lio]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[rviz-3]: started with pid [10659]
^C[rviz-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Nationarity, 2022-09-06:
Add livox_ros_driver (CustomMsg.h and CustomPoint.h) in ~/catkin_ws/src/FAST_LIO/include.

Again.
'~/catkin_ws$ catkin_make

Previously, there were several fatal errors: here
[100%] Built target fastlio_mapping.

roslaunch fast_lio mapping_avia.launch
 .... .ERROR: cannot launch node of type... is no longer occurring.

closed 2022-09-06 by Nationarity