hku-mars/FAST_LIO #190: 有关室内效果的问题

hr2894235132, 2022-10-16:
您好，首先非常感谢您优秀的工作！！！
我想请问一下，我在用mid70+外置imu测试算法的时候，录制了一个室内的bag数据，每次跑到最后30秒左右的时候终端就会出现“no effective point”的警告，然后rviz的表现就是里程计“飞”了。imu和雷达已经根据另一个非常优秀的工作lidar_imu_init进行了同步。
我不太明白为什么会出现这个问题，可以抽空帮我解答一下嘛，非常感谢！！！

zfc-zfc, 2022-10-18:
> 您好，首先非常感谢您优秀的工作！！！ 我想请问一下，我在用mid70+外置imu测试算法的时候，录制了一个室内的bag数据，每次跑到最后30秒左右的时候终端就会出现“no effective point”的警告，然后rviz的表现就是里程计“飞”了。imu和雷达已经根据另一个非常优秀的工作lidar_imu_init进行了同步。 我不太明白为什么会出现这个问题，可以抽空帮我解答一下嘛，非常感谢！！！

是否雷达被遮挡？可以检查一下出现“no effective point”的那一帧点云的size，很可能是该帧没有点 导致里程计漂移

Neal-zz, 2022-10-19:
> 您好，首先非常感谢您优秀的工作！！！ 我想请问一下，我在用mid70+外置imu测试算法的时候，录制了一个室内的bag数据，每次跑到最后30秒左右的时候终端就会出现“no effective point”的警告，然后rviz的表现就是里程计“飞”了。imu和雷达已经根据另一个非常优秀的工作lidar_imu_init进行了同步。 我不太明白为什么会出现这个问题，可以抽空帮我解答一下嘛，非常感谢！！！

您好，想请教您两个问题：1. 你的IMU选购的是哪一款。2. 你是如何实现IMU和mid-70的硬件同步的？

hr2894235132, 2022-10-20:
> > 您好，首先非常感谢您优秀的工作！！！ 我想请问一下，我在用mid70+外置imu测试算法的时候，录制了一个室内的bag数据，每次跑到最后30秒左右的时候终端就会出现“no effective point”的警告，然后rviz的表现就是里程计“飞”了。imu和雷达已经根据另一个非常优秀的工作lidar_imu_init进行了同步。 我不太明白为什么会出现这个问题，可以抽空帮我解答一下嘛，非常感谢！！！
> 
> 您好，想请教您两个问题：1. 你的IMU选购的是哪一款。2. 你是如何实现IMU和mid-70的硬件同步的？
您好！最开始选择的imu效果不是很好，最新用的是realsence内置的imu，关于同步我只做了雷达的时间同步，然后分别对雷达imu以及相机雷达做了标定。

closed 2023-01-01 by stale[bot]

wyb17js, 2023-12-20:
I meet the same question with you, have you solve this?