hku-mars/FAST_LIO #192: a sliding window with fast-lio2

jkff00, 2022-11-01:
Dear Author：
          Hello! Many front ends of slam systems use a sliding window approach to optimize the system state frame by frame. 
Every time a new frame of point cloud is acquired, all the states in the sliding window will be optimized once. Until the most marginal state is separated from the sliding window, the corresponding pose of the state will be output as the result of the front odometer.Therefore, I would like to add sliding window approach to fast_lio2, does fast_lio2 allow it? 
         Thank you very much!

jkff00, 2023-01-01:
Hello Author.
  Thank you very much for your reply! We have temporarily solved the global optimization problem by reproducing the relevant paper  "Large-Scale LiDAR Consistent Mapping using Hierarchical LiDAR Bundle Adjustment" from your lab. Very much looking forward to your subsequent work.

  Jin Xiaofeng

WINNER 东区
***@***.***

抄送: "WINNER ***@***.******@***.***&gt;;

closed 2023-01-23 by stale[bot]

chenxiaocongAI, 2024-08-04:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

兄弟，你能从ikdtree构建的地图删除历史点吗？还是你基于其他kdtree地图？