hku-mars/FAST_LIO #199: 雷达time相关的问题

chenjianqiang199526, 2022-11-19:
你好，我用的是镭神的16线雷达录制的雷达数据，我看了一下数据的格式跟velodyne的格式是一样的都是x y z instensity ring time ,因此我用velodyne.yaml的参数，但是为什么一直报的Failed to find match for field 'time'

closed 2023-01-01 by stale[bot]

PigletPh, 2023-03-04:
@chenjianqiang199526 你好，这个问题解决了吗

TangSZk, 2023-03-10:
更改ros包发布话题之前每个点的数据格式使用fastlio定义的格式（见preprocess.h中对应的格式）或在fastlio中增加你所用数据类型的数据结构

shahaizi0, 2023-03-10:
更改ros包发布话题之前每个点的数据格式使用fastlio定义的格式（见preprocess.h中对应的格式）或在fastlio中增加你所用数据类型的数据结构

PigletPh, 2023-03-30:
@TangSZk 你好，我对比了下FAST-LIO中velodyne雷达定义格式，发现是一样的，但我将time的类型从double转为float，编译没有错但是雷达数据无法导入到，请问怎么回事，万分感谢
[image]

shaoxh, 2024-02-05:
Make sure that field 'time' is really in your the ROS topic.
Got a similar here: https://github.com/hku-mars/FAST_LIO/issues/302

lijing137, 2024-04-20:
> @TangSZk 你好，我对比了下FAST-LIO中velodyne雷达定义格式，发现是一样的，但我将time的类型从double转为float，编译没有错但是雷达数据无法导入到，请问怎么回事，万分感谢 [image]

You can try using the latest LiDAR driver（my lidar is leishenC32）, I solved the “Failed to find match for field time'' issue this way.