hku-mars/FAST_LIO #221: using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t

Rotoslider, 2023-03-12:
I get a warning when building Fast_Lio
`warning: ‘using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t [-Wdeprecated-declarations]
   52 |     (uint16_t, ring, ring)

pcl_macros.h:94:9: note: declared here
   94 |   using uint16_t [[deprecated("use std::uint16_t instead of pcl::uint16_t")]] = std::uint16_t;`

using pcl 1.10. Ubuntu 20.04 on a Jetson AGX. 
While running a bag file I see the error in yellow:
`**Failed to find match for field 'ring'**.
[ mapping ]: time: IMU + Map + Input Downsample: 0.006932 ave match: 0.059468 ave solve: 0.002559  ave ICP: 0.064655  map incre: 0.016817 ave total: 0.088404 icp: 0.062094 construct H: 0.000524 
**Failed to find match for field 'ring'.**
[ mapping ]: time: IMU + Map + Input Downsample: 0.004389 ave match: 0.053956 ave solve: 0.002595  ave ICP: 0.051593  map incre: 0.012388 ave total: 0.078387 icp: 0.056624 construct H: 0.000659 
**Failed to find match for field 'ring'.**
[ mapping ]: time: IMU + Map + Input Downsample: 0.004147 ave match: 0.052604 ave solve: 0.002456  ave ICP: 0.052583  map incre: 0.016472 ave total: 0.076659 icp: 0.055136 construct H: 0.000641 
`
This does not seem to affect the process but it is a concern.

closed 2023-04-03 by stale[bot]

Rotoslider, 2023-04-03:
**Failed to find match for field 'ring'**
It seems this error is for a velodyne laser but I am using Ouster. What are the details of the reason behind the error and what is the correction.

Carl-1999, 2024-03-05:
I have the same question.Could you please tell me if you solve the problem? Thank you very much!

Rotoslider, 2024-09-22:
I  would mark this closed since it did fix it for me but for some reason the code in this repository has not been updated.  So if anyone downloads now its still broke.

juliangaal, 2024-10-30:
Updated another PR #365 that includes fixes to the config file ouster in ros2 as well @doctorcolossus