hku-mars/FAST_LIO #224: How to use with Livox mid360

vanstrong12138, 2023-03-19:
mid360 can only use Livox-SDK2 and livox_ros_driver2.

XW-HKU, 2023-03-19:
just use it

closed 2023-03-19 by XW-HKU

vanstrong12138, 2023-03-20:
> just use it

[image]
[image]
[image]
[image]
my livox_ros_driver2 has already built in my '/home/d_livox/src/'  ,but in the'/home/fastlio2',  i do 'source ~/d_livox/devel/setup.bash'.  no error.  but when i 'catkin_make' . it is error

geniusdo, 2023-04-18:
just put both livox_ros_driver and livox_ros_driver2 in your src folder