hku-mars/FAST_LIO #234: Support ROS2

Ericsii, 2023-04-06:
Is any plan to support ROS2? I'm porting FAST_LIO into ROS2-humble. Wondering if you have a plan to support ROS2. If so could you please create a new branch `ros2`? I'll take few PRs to porting my ros2 code to this branch

Ecstasy-EC, 2023-04-08:
Absolutely yes! Our team has been busy these days. We will create the branch for you in the coming week.

Ericsii, 2023-04-09:
Very appreciate. I made a simple port of FAST_LIO to ros2 version. Please feel free to check it on my fork. https://github.com/Ericsii/FAST_LIO

Ecstasy-EC, 2023-04-10:
> Very appreciate. I made a simple port of FAST_LIO to ros2 version. Please feel free to check it on my fork. https://github.com/Ericsii/FAST_LIO

Thanks! We will reach you as soon as we finish the test.

Ecstasy-EC, 2023-04-20:
@Ericsii   Could you please provide me with your email for communicating more efficiently about some problems I met when testing your branch?

TheodorPatrickZ, 2023-05-17:
> Very appreciate. I made a simple port of FAST_LIO to ros2 version. Please feel free to check it on my fork. [https://github.com/Ericsii/FAST_LIO ](https://github.com/Ericsii/FAST_LIO)

Hi are you able to run the provided rosbags when convertet to ros2?
It is not working for me.

Ecstasy-EC, 2023-05-17:
> > Very appreciate. I made a simple port of FAST_LIO to ros2 version. Please feel free to check it on my fork. [https://github.com/Ericsii/FAST_LIO ](https://github.com/Ericsii/FAST_LIO)
> 
> Hi are you able to run the provided rosbags when convertet to ros2? It is not working for me.

Hi. I also tried this repo but I could not run successfully yet. I was trying to contact the author but got no response.

Ericsii, 2023-05-19:
@Ecstasy-EC Sorry for the late reply. Here is my email ericfengx@foxmail.com.

Ericsii, 2023-05-19:
> > Very appreciate. I made a simple port of FAST_LIO to ros2 version. Please feel free to check it on my fork. [https://github.com/Ericsii/FAST_LIO ](https://github.com/Ericsii/FAST_LIO)
> 
> Hi are you able to run the provided rosbags when convertet to ros2? It is not working for me.

rosbags for ros1 is not compatible in ros2. You can follow this tutorial to convert rosbags to ros2. https://ternaris.gitlab.io/rosbags/topics/convert.html

Ericsii, 2023-05-22:
@Ecstasy-EC Can you reopen this issue? So that others can communicate about ros2 migration.

Ecstasy-EC, 2023-05-23:
> @Ecstasy-EC Can you reopen this issue? So that others can communicate about ros2 migration.

Reopened.

andy-zhuo-02, 2023-05-25:
I'm also interested in the ROS2 version of FAST-LIO. Is there a stable version yet?

andy-zhuo-02, 2023-05-25:
I have successfully run through the data stream, but there seems to be a problem with the results of the mapping. The appearance of point clouds is very strange. The range of point clouds is very small and their positioning is easy to drift. I don't know if there are any suggestions.
[image]
[image]

andy-zhuo-02, 2023-05-26:
> > I have successfully run through the data stream, but there seems to be a problem with the results of the mapping. The appearance of point clouds is very strange. The range of point clouds is very small and their positioning is easy to drift. I don't know if there are any suggestions. [image] [image]
> 
> Are you able to run FAST-LIO2 in ros1 using the same config on the data?

It seems that, YES.

here is the fast-lio in ROS noetic. I compared all config in mid70.yaml, or I missed something?

[image]

Ecstasy-EC, 2023-05-26:
@Ericsii  Would you please provide some help if you have any ideas?

Ericsii, 2023-05-30:
> Are you able to run FAST-LIO2 in ros1 using the same config on the data?

Can you give me your configuration file? This may be because `publish.dense_publish_en` is set to `false`.

Ericsii, 2023-06-23:
Please keep this issue open.

Taeyoung96, 2023-07-26:
@Ecstasy-EC @Ericsii   

I got the FAST-LIO2 ros2 version up and running in docker based on the ros2 branch created by Ericsii and it works pretty well!  

The dockerized repository is shown below. 
I even added a function to save the trajectory and 3D map.

After converting the M2DGR dataset to a ros2 bagfile, I performed experiments using the velodyne and handsfree imu.

See below for more details,
https://github.com/Taeyoung96/FAST_LIO_ROS2/tree/ros2

[image]

ryanpdoc57, 2023-11-02:
Hello! I was able to successfully fun ROS2 version of fast-lio in Humble, but am having issues building the package using ROS2 foxy. Has anyone else experienced this?

ryanpdoc57, 2023-11-02:
@Ericsii Hello! I was able to successfully fun ROS2 version of fast-lio in Humble, but am having issues building the package using ROS2 foxy. Have you tested this in foxy? If so did you run into any build issues? It is specifically with a service and a service callback from the lasermapping.cpp file.

Ericsii, 2023-11-24:
> @Ericsii Hello! I was able to successfully run ROS2 version of fast-lio in Humble, but am having issues building the package using ROS2 foxy. Have you tested this in foxy? If so did you run into any build issues? It is specifically with a service and a service callback from the lasermapping.cpp file.

Sorry, I didn't test this on ROS Foxy. Can you provide a build log on foxy?

Ericsii, 2023-11-24:
@Ecstasy-EC Perhaps we can now create a new branch for ROS2?

wyb17js, 2023-12-21:
Hello, we use this transplanted fast-lio2 algorithm to realize the robot's positioning function on humble, but we encounter a problem: the positioning is lost after a period of time. We find that: the time that the positioning can be maintained is related to the radar release frequency, the higher the frequency, the longer the duration; With the operation of the program, there will be an interval in the use of radar data. For example, I am publishing a point cloud at a frequency of 10HZ. Originally, my radar data is matched with imu data every 0.1s and used by the algorithm, but with the operation of the program, my radar data may be used at an interval of 0.2s and 0.3s, and this interval gradually increases. When the interval reaches 1s, the location is lost. I was wondering if you had the same problem.Do you have any good suggestions on this matter？
I'm looking forward to your reply.

Ericsii, 2023-12-21:
> For example, I am publishing a point cloud at a frequency of 10HZ. Originally, my radar data is matched with imu data every 0.1s and used by the algorithm, but with the operation of the program, my [...]

@wyb17js have you tested this on the original version of FAST_LIO for ROS1? If the result is the same, you may check whether the timestamps of IMU and lidar are aligned.

wyb17js, 2023-12-21:
Not yet, but I have tested it in a simulation environment, using radar and imu plugins, and there is also a loss of location, but the location lasts longer than it actually does. So how do I align the radar and imu timestamps? The radar I use is mid360, no external imu

Ericsii, 2023-12-21:
> I have tested it in a simulation environment

Which simulator did you use? I have tested on Gazebo the lidar output frequency is not usable for common SLAM algorithms. Please use real environment

wyb17js, 2023-12-21:
> > I have tested it in a simulation environment
> 
> Which simulator did you use? I have tested on Gazebo the lidar output frequency is not usable for common SLAM algorithms. Please use real environment

I use Gazebo,too. Does time synchronization refer to hardware synchronization? I noticed that there is code in the program to handle timestamp deviation, and it takes effect when time_sync_en is turned on. I tried this feature but it didn't work well.

Ericsii, 2023-12-22:
> I use Gazebo,too. Does time synchronization refer to hardware synchronization? I noticed that there is code in the program to handle timestamp deviation, and it takes effect when time_sync_en is [...]

Please test it in a real-world environment. The online time synchronization algorithm only estimates the fixed timestamp offset between lidar and IMU. For dynamic time offset, it may not work as expected

wyb17js, 2023-12-22:
> Please test it in a real-world environment. The online time synchronization algorithm only estimates the fixed timestamp offset between lidar and IMU. For dynamic time offset, it may not work as [...]

So how do I deal with dynamic time offset? I don't know much about it. Could you give me some advice?

Ericsii, 2023-12-28:
> So how do I deal with dynamic time offset? I don't know much about it. Could you give me some advice?

Sorry, I'm not an expert in timestamp calibration. You can search for some academic papers in these related fields.

Ecstasy-EC, 2024-02-25:
The issue is closed. A new branch for ROS2 has been created and the pull request from @Ericsii is merged. Please open new issues if any other problems are found in the ROS2 version.

closed 2024-02-25 by Ecstasy-EC

lingxe01, 2024-07-15:
Have you tested it on ros version foxy? I got an error when compiling.
`/opt/ros/foxy/include/rclcpp/create_service.hpp:43:3: error: no matching function for call to ‘rclcpp::AnyServiceCallback<std_srvs::srv::Trigger>::set(std::_Bind<void (LaserMappingNode::*(LaserMappingNode*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<const std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >)>)’
   43 |   any_service_callback.set(std::forward<CallbackT>(callback));`

wendwosenbb, 2024-07-23:
faced same issue on ros2 galactic.

wendwosenbb, 2024-07-23:
looks like a third place holder is required for the service request header/id. 
changing line 945 of lasermapping.cpp to
`        map_save_srv_ = this->create_service<std_srvs::srv::Trigger>("map_save", std::bind(&LaserMappingNode::map_save_callback, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));`

and line 1115 to
`    void map_save_callback(  const std::shared_ptr<rmw_request_id_t> request_header,
  const std::shared_ptr<std_srvs::srv::Trigger::Request> req,
  const std::shared_ptr<std_srvs::srv::Trigger::Response> res)`
  
 solved the error
 `/opt/ros/galactic/include/rclcpp/create_service.hpp:43:3: error: no matching function for call to ‘rclcpp::AnyServiceCallback<std_srvs::srv::Trigger>::set(std::_Bind<void (LaserMappingNode::*(LaserMappingNode*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<const std_srvs::srv::Trigger_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > >)>)’ 43 |   any_service_callback.set(std::forward<CallbackT>(callback));`

Ericsii, 2024-07-23:
> Have you tested it on ros version foxy? I got an error when compiling. `/opt/ros/foxy/include/rclcpp/create_service.hpp:43:3: error: no matching function for call to [...]

You can use this fork's version it solve this problem in foxy https://github.com/hku-mars/FAST_LIO/pull/334