hku-mars/FAST_LIO #247: Does physical orientation of Livox lidar affects SLAM?

rahulsharma11, 2023-05-13:
Hi.

I have Livox MID360 lidar. I am trying to do small room indoor slam with Fastlio.
While running the Fastlio sometime it gets drift while entering in small rooms.

I have placed the lidar facing in front direction so that its FOV covers the full vertical direction.
I just want to know does placing the LIDAR Horizontal/vertical physically affects the Fastlio SLAM?

Is there any mandatory orientation of Lidar is recommendate?

Thanks.

snakehaihai, 2023-05-14:
LIDAR Horizontal/vertical physically affects the Fastlio SLAM? Yes it affects.
Make sure that you can see at least 3 different orientations of surface normal.  Example would be, If you place it vertically, and pass through a corridor. it only can see 2 different orientations. and thus it will drift in front and back direction.

rahulsharma11, 2023-05-14:
@snakehaihai , Thanks for reply.

I am planning to use 2 mid360. As in corridor area, it is difficult to continue with only one lidar.

closed 2023-06-09 by stale[bot]