hku-mars/FAST_LIO #253: Questions about the IMU data of Livox Avia Rosbag

fanggenzaiXHBS, 2023-06-21:
Hello, author, I am very sorry to bother you.
I downloaded the **Livox Avia Rosbag** you provided and found that the acceleration value of the IMU is basically close to 0 even when it is stationary, did you take gravity out of the IMU output。

XW-HKU, 2023-06-21:
its unit is G（i.e. 9.8m/s2）

Joosoo1, 2023-06-23:
> its unit is G（i.e. 9.8m/s2）

@XW-HKU    Can make the acceleration unit of imu output from livox in meters per second? If so, please let me know, thanks!

closed 2023-07-15 by stale[bot]

daijicheng, 2023-12-14:
> > its unit is G（i.e. 9.8m/s2）
> 
> @XW-HKU Can make the acceleration unit of imu output from livox in meters per second? If so, please tell me, thanks!

I think the code in https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE?tab=readme-ov-file will help you.