hku-mars/FAST_LIO #256: Velodyne point cloud visualization

jeho-lee, 2023-07-01:
Hi. While I am trying to use FAST LIO from the ROS bag containing VLP-16 point clouds and IMU data, the odometry seems correct and consistent, but the point cloud is not shown in Rviz.

Below are the Rviz and velodyne.yaml configuration.

[image]

[image]

I've tried to change some Rviz configurations (e.g., currPoints Topic to "/velodyne_points"), but didn't work.
Did I miss something to modify in load_livox.rviz configuration file? Any help would be thankful.

lxystar00, 2023-07-11:
Hi, I have the same question, could i see your terminal? My terminal give me a warn 'No point, skip this scan!'. Thank you!!

closed 2023-08-03 by stale[bot]

maffan-96, 2024-03-07:
> Hi, I have the same question, could i see your terminal? My terminal give me a warn 'No point, skip this scan!'. Thank you!!

Hey, were you able to resolve the issue?

shaoxiang, 2024-07-09:
> > Hi, I have the same question, could i see your terminal? My terminal give me a warn 'No point, skip this scan!'. Thank you!!嗨，我有同样的问题，我能看到你的终端吗？我的终端给我一个警告“没有意义，跳过此扫描！谢谢！！
> 
> Hey, were you able to resolve the issue?嘿，你能解决这个问题吗？
Refer to #278 , please check the timestamps of IMU and Lidar

yucong-zhang, 2024-12-22:
I'm having this problem with UrbanNav, any solutions?