hku-mars/FAST_LIO #260: 激光和IMU的时间同步参数如何配置

5204338, 2023-07-19:
common:
    lid_topic:  "/livox/lidar"
    imu_topic:  "/wit/imu"
    time_sync_en: true         # ONLY turn on when external time synchronization is really not possible
    time_offset_lidar_to_imu: 1689742965.1697087288 
时间同步参数：time_offset_lidar_to_imu 的配置？
我现在设置该参数的方法是，每次激光雷达开启后，都配置一次，有什么一劳永逸，永久的配置该参数的方法吗？

closed 2023-08-18 by stale[bot]

gc625-kodifly, 2024-03-19:
Hi, based on your imu topic, are you using the 10 axis imu from yahboom? if so, do you have any suggestions on the parameter values for acc_cov, gyr_cov, b_acc_cov, b_gyr_cov?

WN-Wolf, 2024-07-12:
@5204338 你好，我也使用的是mid-70外加wit IMU，我通过LiDAR_IMU_Init得到一个txt文档
[image]
但是，我如何将txt中文档的内容对应t填写在config/avia.yaml文件中呢？
[image]