hku-mars/FAST_LIO #262: 为什么室内工厂建图的出发点和停止点对应不上

5204338, 2023-07-26:
建图的起点和终点在现实中是同一个点，但是在地图中生成的起点和终点对应不上，这是为什么呢？

5204338, 2023-07-26:
这里是我的建图效果

[image]

5204338, 2023-07-26:
第一个情况是：在最后的转角，平面转向向下倾斜了
第二个情况是：第一条长直线和第二条短直线长度对不上

linjie111, 2023-08-07:
你解决了吗？

5204338, 2023-08-22:
> 你解决了吗？

没有呢

NHMMing, 2023-08-28:
> 需要根据激光雷达的实际情况和场景对应地调整一下参数，我们的默认参数只能handle大部分的case，并不是能处理所有的问题

参数是指filter_size_surf、filter_size_map、point_filter_num、blind、acc_cov、gyr_cov、b_acc_cov、b_gyr_cov这些吗？
此外，我在自己的实验中也有两个问题没弄明白，还希望您帮我解答下：
1. 本仓库代码中acc_cov、gyr_cov、b_acc_cov、b_gyr_cov的单位是什么？因为我换了一个自己的imu，我要做一些参数适配，这几个值我暂时没找到代码里用的单位是啥
2. 我在室外环境中测试时，轨迹也有上述相似的情况，在起点终点无法闭环，甚至相差很远。在一些转弯处，heading角差的也比较多。我使用的不是avia雷达，用的hap，相比而言垂直视场角小了很多，我可以调整哪些参数来让轨迹更优？并且我的雷达不是正视放置的，在前左上坐标系中，我的雷达安装角度在z轴有个逆时针30度的偏角。
以上两个问题还希望您有空帮忙解答下

NHMMing, 2023-08-28:
> 需要根据激光雷达的实际情况和场景对应地调整一下参数，我们的默认参数只能handle大部分的case，并不是能处理所有的问题

轨迹如图，绿色为真值，红色的实际跑出来的
[image]

closed 2023-09-21 by stale[bot]

silent-zhou, 2024-01-04:
> 参数是指filter_size_surf、filter_size_map、point_filter_num、blind、acc_cov、gyr_cov、b_acc_cov、b_gyr_cov这些吗？ 此外，我在自己的实验中也有两个问题没弄明白，还希望您帮我解答下：
>
> 1. [...]
你好，请问acc_cov、gyr_cov、b_acc_cov、b_gyr_cov如何做自己的IMU适配呢

silent-zhou, 2024-07-09:
> 2\. 我使用的不是avia雷达，用的hap，相比而言垂直视场角小了很多，我可以调整哪些参数来让轨迹更优？并且我的雷达

请问你解决了吗，我也有相似的疑惑，这些噪声参数和inti_P有什么标准呢