hku-mars/FAST_LIO #264: require understanding of IMU params in config file

7DoF, 2023-07-31:
Hi, Thanks for sharing your work. 

As i was going through the code I faced difficulty in understanding what these IMU params [in the yaml files under the config folder]  mean:

```
# acc_cov: 0.1
# gyr_cov: 0.1
# b_acc_cov: 0.0001
# b_gyr_cov: 0.0001
```

my hardware setup includes a Ouster 128 lidar with QC RB5 dev-board. I'm using the internal IMU of the RB5 board. 

I've used imu_utils for IMU intrinsic calculations.
the different values [intrinsic] it gives are:
for accel:

[image]

for gyro:
[image]

it also generates a .yaml file with averaged noise params for the commonly used terms:
```
%YAML:1.0
---
type: IMU
name: rb5imu
Gyr:
   unit: "gyr_n: rad / sqrt(s), gyr_w: rad / s^2 / sqrt(Hz)"
   avg-axis:
      gyr_n: 5.4709957607599371e-05
      gyr_w: 1.5816674386101611e-06
   x-axis:
      gyr_n: 5.0324793946038370e-05
      gyr_w: 1.7560913290061533e-06
   y-axis:
      gyr_n: 5.4284409696253083e-05
      gyr_w: 7.4827664885237793e-07
   z-axis:
      gyr_n: 5.9520669180506661e-05
      gyr_w: 2.2406343379719522e-06
Acc:
   unit: "acc_n: m / s^2 / sqrt(Hz), acc_w: m / s^3 / sqrt(Hz)"
   avg-axis:
      acc_n: 9.2035905166550178e-04
      acc_w: 1.3220037181370156e-05
   x-axis:
      acc_n: 7.5207169864227694e-04
      acc_w: 4.6902506183408122e-06
   y-axis:
      acc_n: 1.2724660685080129e-03
      acc_w: 3.4331061789856015e-05
   z-axis:
      acc_n: 7.3653938784621568e-04
      acc_w: 6.3879913591364199e-07
```

In your IMU_processing.hpp code, I found the following relevent information:
```
void ImuProcess::set_gyr_cov(const V3D &scaler)

{
  cov_gyr_scale = scaler;
}

void ImuProcess::set_acc_cov(const V3D &scaler)
{
  cov_acc_scale = scaler;
}

void ImuProcess::set_gyr_bias_cov(const V3D &b_g)
{
  cov_bias_gyr = b_g;
}

void ImuProcess::set_acc_bias_cov(const V3D &b_a)
{
  cov_bias_acc = b_a;
}
```

From what I understand, the error term **Scale factor** and **bias instability** are being used here [in the above written code].

However, I am not sure I'm utilizing the error term [from imu_utils ] in the yaml config file. Correct me if I'm wrong please.

Thanks,

NHMMing, 2023-08-21:
我也有相同的疑问，代码中在使用
# acc_cov: 0.1
# gyr_cov: 0.1
# b_acc_cov: 0.0001
# b_gyr_cov: 0.0001
四个参数时，它们的单位分别是什么？此外，这四个参数如果与实际有较大偏差，会影响里程计的精度吗？

closed 2023-09-13 by stale[bot]

mahmood-ul-hasan, 2024-09-17:
I have same problem