hku-mars/FAST_LIO #266: Gravity aligned position estimation

Pallav1299, 2023-08-01:
Hi. Kudos to the great work.
I noticed that there is no gravity alignment in the initialization step which accounts for the surface tilt. This means if the ground surface is not flat, the origin frame would initialize abnormally. 
For example: If this isn't handled the estimated pose might gain/reduce altitude while moving along x or y axes. How do you suggest to handle the same?
Correct me if I'm wrong. @XW-HKU

Giannis-Alamanos, 2023-08-22:
Hello. 
You're absolutely correct. You can estimate the initial tilt through the accelerometer of the IMU and adjust the pose accordingly.

Pallav1299, 2023-08-22:
@Giannis-Alamanos, this could help well in static initialization situations. What problems do you expect with initializing during motions? Do you know how we can adapt to such situations?

Giannis-Alamanos, 2023-08-30:
@Pallav1299 to initialize during motion I would utilize the acceleration information from both Odometry and IMU's accelerometer. However if you initialize statically once in the beginning the estimated pose will not gain/reduce altitude while moving along x or y axes.

satyajitghana, 2023-09-14:
```
    Eigen::Quaterniond rotation = Eigen::Quaterniond::FromTwoVectors(mean_acc, Eigen::Vector3d::UnitZ());
    mean_acc = rotation * mean_acc;
    mean_gyr = rotation * mean_gyr;
    state_ikfom init_state = kf_state.get_x();
    init_state.rot = rotation;
    init_state.grav = S2(-mean_acc / mean_acc.norm() * G_m_s2);
```

closed 2023-10-07 by stale[bot]