hku-mars/FAST_LIO #273: bug: the scale of the exponential value is zero

lucabart97, 2023-09-22:
## Description
In the code, the scale is set to ```scalar_type(1/2)``` when the exponential value is used. This is to ensure that there is a compile-time conversion between the scalar type of the class and the scalar type of the exponential value. However, this can cause an issue because **the value is computed at compile time with integer division, which result zero.**

## Position
You can find this issue 9 times in the code by searching for ```scalar_type(1/2)```. The issue is ever present in the exponential computation.

## Small test
```
#include <iostream>
template <class scale_type> class Test {
public:
  Test() { std::cout << scale_type(1 / 2) << ' '; }
};

int main() {
  Test<int> t0;
  Test<float> t1;
  Test<double> t2;
  return 0;
}
```

Result:
```
0 0 0
```

lucabart97, 2023-10-13:
ping

lucabart97, 2023-11-04:
Last ping

closed 2023-11-06 by stale[bot]

xiaodong2077, 2024-03-18:
Agreed.

mulin33, 2026-10-08:
Agreed +1