hku-mars/FAST_LIO #278: No point,skip this scan!  when using realsense d435i IMU

HowieHepburn, 2023-10-07:
lidar: mid-70
imu: ralsense d435i  (  
<arg name="enable_gyro"         default="true"/>
<arg name="enable_accel"        default="true"/>

  <arg name="unite_imu_method"          default="copy"/>
  
)

root@howiePC:/home/howie# roslaunch fast_lio mapping_avia.launch 
... logging to /root/.ros/log/879f2006-6517-11ee-80af-ab28e3a17af7/roslaunch-howiePC-51092.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://howiePC:44117/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /camera/imu
 * /common/lid_topic: /livox/lidar
 * /common/time_offset_lidar_to_imu: 0.0
 * /common/time_sync_en: True
 * /cube_side_length: 1000.0
 * /feature_extract_enable: False
 * /filter_size_map: 0.5
 * /filter_size_surf: 0.5
 * /mapping/acc_cov: 0.1
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 450.0
 * /mapping/extrinsic_R: [-0.001962, -0.99...
 * /mapping/extrinsic_T: [0.037528, 0.1064...
 * /mapping/extrinsic_est_en: False
 * /mapping/fov_degree: 90
 * /mapping/gyr_cov: 0.1
 * /max_iteration: 3
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: True
 * /point_filter_num: 3
 * /preprocess/blind: 0.2
 * /preprocess/lidar_type: 1
 * /preprocess/scan_line: 6
 * /publish/dense_publish_en: True
 * /publish/path_en: False
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /runtime_pos_log_enable: False

NODES
  /
    laserMapping (fast_lio/fastlio_mapping)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[laserMapping-1]: started with pid [51100]
process[rviz-2]: started with pid [51101]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Multi thread started 
p_pre->lidar_type 1
~~~~/home/howie/catkin_ws/src/FAST_LIO/ file opened
Self sync IMU and LiDAR, time diff is -1696683450.2298572063 
[ WARN] [1696686238.128496571]: No point, skip this scan!

[ WARN] [1696686238.226635563]: No point, skip this scan!

 * /common/time_sync_en: True
 *  * /common/imu_topic: /camera/imu
 * /common/lid_topic: /livox/lidar
 topic /camera/imu and /livox/lidar has messages.
 when i roslaunch realsense2_camera rs_camera.launch, warnings shows like this:
   [ INFO] [1696686223.818603063]: Setting Dynamic reconfig parameters.
  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)

  [ WARN] [1696686228.853362730]: 
[ WARN] [1696686228.853484775]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 07/10 21:43:48,987 WARNING [140021994010368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1696686229.038353212]: SELECTED BASE:Depth, 0
[ INFO] [1696686229.048308416]: RealSense Node Is Up!
 07/10 21:43:49,172 WARNING [140021994010368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 

 **rostopic echo /camera/imu** 
 
   x: -0.026613891124725342
  y: -9.818659782409668
  z: 0.7886490821838379
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
^Cheader: 
  seq: 103983
  stamp: 
    secs: 1696686935
    nsecs: 331386566
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 4.611313670466188e-06
  y: -0.005240695551037788
  z: 0.0017354879528284073
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.06632091104984283
  y: -9.837021827697754
  z: 0.8502249717712402
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---

**rostopic echo /camera/imu** 

    z: 0.2849999964237213
    reflectivity: 0
    tag: 0
    line: 0
  - 
    offset_time: 99800000
    x: 0.8920000195503235
    y: 0.5049999952316284
    z: 0.2849999964237213
    reflectivity: 0
    tag: 0
    line: 0
  - 
    offset_time: 99810000
    x: 0.9089999794960022
    y: 0.5149999856948853
    z: 0.2930000126361847
    reflectivity: 0
    tag: 1
    line: 0
  - 
    offset_time: 99820000
    x: 0.9100000262260437
    y: 0.5170000195503235
    z: 0.29499998688697815
    reflectivity: 0
    tag: 0
    line: 0
  - 
    offset_time: 99830000
    x: 0.9110000133514404
    y: 0.5180000066757202
    z: 0.296999990940094
    reflectivity: 0
    tag: 0
    line: 0
---

HowieHepburn, 2023-10-07:
addition:
1: it seems that the warning of rs_camera doesn't matter.  
 https://blog.csdn.net/sinat_16643223/article/details/115289206

2: rs_camera args 
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="unite_imu_method"          default="copy"/>

closed 2023-10-30 by stale[bot]

panchenwei, 2024-05-15:
I also get this problem.
Are you slove it?

panchenwei, 2024-05-15:
@Chandan-kumar-R

Chandan-kumar-R, 2024-05-16:
@panchenwei I think its due to time synchronization. As stated in the Readme file, it’s crucial to synchronize your IMU and LIDAR data. In my specific case, I was utilizing a Livox LIDAR with a custom point cloud. To address this, I adjusted the timestamp of the point cloud data to match the ROS timestamp, which was already being used for my IMU data. This ensured that both sets of data were synchronized effectively.

The thing you can do is to write a simple ros node to compare the time stamp of both imu and lidar.

HowieHepburn, 2024-05-17:
@panchenwei @panchenwei You are right, it is indeed because of the time synchronization problem. At that time, I simply opened the software time synchronization of the LIO system to obtain the time difference, and then set the time difference parameters. But this way I have to reset the parameters every time, can you share Check your ros node?